easyw/kicadStepUpMod #2: Doesn't extrude multiple boards

morganrallen, 2018-05-16:
Thanks for this great plugin, it's really help push me to learn FreeCAD better.

I'm working on a board that consists of multiple `Edge.Cuts` and only a single one appears in `Boards_Geoms`

I tested this with KiCAD 5.0.0-rc2-dev-unknown-554cb54~64~ubuntu17.10.1
A blank board with two squares is good enough to test.

[image]
[image]

easyw, 2018-05-16:
Hi @morganrallen 
KiCAD doesn't support multiple separated 'Edge.Cuts' board on a single .kicad_pcb file, so StepUp doesn't either.

What you could do is:
* make a single bigger 'Edge.Cut' and add 'V-scoring' to your pcb
* make a single bigger 'Edge.Cut' and add 'milling' and 'mouse-bites' to your pcb
* mix milling and scoring in your pcb

Some related discussion at KiCAD forum or from a manufacturer here:
1) https://forum.kicad.info/t/v-scoring-of-a-board/7237
2) https://forum.kicad.info/t/panelising-internal-edge-cuts/164
3) https://www.eurocircuits.com/panel-instructions-vcut-basics/

PS: I noticed you are using FC0.16... please consider to update it to FC0.17 which is the new stable release and it is much featured for mechanical integration.

morganrallen, 2018-05-16:
Interesting, weather it _supports_ it or not, it does work. I've been making multipart boards for a little while now. Here's one example (don't have the source immediately at hand, here's OSHParks rendering)

[image]

When I first made this board KiCAD failed to render all 3 parts in the 3d viewer but it still worked. In the current version of KiCAD multiple part *do* render now and things like push-pull routing honor `Edge.Cuts` no matter how many there are. Here's the test piece from earlier rendered in KiCAD.

[image]

So, seems to support it just fine to me, unless there is something I'm missing. And personally, this would be pretty disappointing to not support this, as the single purpose to use StepUp was to help make 3D PCBs like this. https://hackaday.io/project/27788-lampster

(I saw 0.17 had been release but as usual Ubuntu is behind, found the PPA and using the newest now)

easyw, 2018-05-18:
Hi,
I've managed to get the full pcb built also when there the board has detached parts... 
[image]
[commit #105](https://github.com/easyw/kicadStepUpMod/commit/6555f0b64f119c3f6ad8a08fa5520b3d3d2c5bb9)

Still I would suggest to use one of the following options:
* add scoring, milling or mouse-bites to your design
* once your pcb is complete, just save a single copy board withing each single detached pcb and do the conversions separately.

morganrallen, 2018-05-20:
Glad to see this was doable. Would it be possible to have it create individual objects for each part? That would then allow fit testing 3d assemblies. Thanks for the improvements so far.

morganrallen, 2018-05-21:
> I see no way to discriminate the outline border from internal cuts (kicad does not either), so there is no straight way to produces detached pcbs.
Yeah, I noticed KiCAD doesn't seem to be able to handle this.

> Anyway you can achieve this by design, creating one file each board or doing a different approach to your design.

That is overly complicated for my needs, what I found works is after importing with StepUp I duplicate the sketch once for each panel part then delete all the other content. A bit slow but it's working.

easyw, 2018-05-21:
Glad it works for you... Everyone have to find his/her way to work with StepUp tools ..
you may also have a try also to push-pull pcb edge between KiCad & FreeCAD... it is a good option to speed up your designs.

morganrallen, 2018-05-21:
That push/pull is my end goal, I would love to be able to select several faces and export them all at once but there is probably going to be some work before that's possible. I appreciate the support.

easyw, 2018-05-21:
> I would love to be able to select several faces and export them all at once but there is probably going to be some work before that's possible.

I miss what you are trying to achieve.. would you please articulate it a bit more? or upload a screencast gif?

easyw, 2018-05-22:
@morganrallen I forgot to quote your name to ping your reply...