easyw/kicadStepUpMod #3: Draw EP pad as circle

jneiva08, 2018-06-26:
Hi,

I was try to verify if a 3d step was matching with the footprint.
When the [footprint](https://github.com/KiCad/kicad-footprints/blob/master/Package_DFN_QFN.pretty/QFN-64-1EP_9x9mm_P0.5mm_EP7.7x7.7mm.kicad_mod) loaded show the EP pad as circle instead a square with rounded corners.

[image]

Now with thermal via.
[image]

I opened this issue only to report.

Thanks, for a usefull addon.

poeschlr, 2018-07-28:
I would guess that the footprint in question had rounded rectangle pads.
Not sure why sometimes rounded rectangle is converted to a circle and sometimes to a rectangle.

As the particular footprint in question no longer exists in the lib, here the link to this footpirnt from the lib tagged as 5.0.0-rc3: https://github.com/KiCad/kicad-footprints/blob/5.0.0-rc3/Package_DFN_QFN.pretty/QFN-64-1EP_9x9mm_P0.5mm_EP7.7x7.7mm.kicad_mod

poeschlr, 2018-07-28:
@easyw would it be easily possible to just interpret rounded rectangle pads as rectangle pads until proper support for rounded rectangle pads is added? (This would allow checking of footprint dimensions for such parts even if it does not fully represent the finer details of the footprint)

easyw, 2018-07-28:
@jneiva08 @poeschlr 
it seems it was not so long to implement 😄 
[image]

Please update the ksu WB (actual release 7.3.3.3).

closed 2018-07-28 by easyw

poeschlr, 2018-07-29:
Wow. Thanks for the update.

jneiva08, 2018-07-29:
Thanks for fix it