easyw/kicadStepUpMod #8: Via drills are missing

diorcety, 2018-12-28:
Adding someting like that at 12675:

```
    for v in mypcb.via:
        m_angle = 0
        if not isinstance(v.drill, (list,)):
             v.drill = [v.drill]
        if 'oval' in str(v.drill).split(',')[0]:  #py3 dict workaround
        #if p.drill.oval:  #maui temp workaround errorchecking
            #sayw(str(p.drill.oval))
            #sayw('drill oval')
            if v.drill[0] >= min_drill_size or p.drill[1] >= min_drill_size:
                xs=v.at[0];ys=-v.at[1]
                #x1=mc.end[0]+m.at[0];y1=-mc.end[1]-m.at[1]
                #radius = float(p.drill[0])/2 #sqrt((xs - x1) ** 2 + (ys - y1) ** 2)
                rx=float(v.drill[0])
                #print (p.drill)
                if len(v.drill)>2:
                    try:
                        #print (p.drill[1])
                        #stop
                        ry=float(v.drill[1])
                    except:
                        ry=rx
                else:
                    ry=rx
                #print(v.at[0],p.at[1], p.drill[0])
                [x1, y1] = rotPoint2([xs, ys], [v.at[0], -v.at[1]], m_angle)
                # sayw('holes solid '+str(holes_solid))
                if holes_solid:
                    obj=createHole3(x1,y1,rx,ry,"oval",totalHeight) #need to be separated instructions   
                else:
                    obj=createHole4(x1,y1,rx,ry,"oval") #need to be separated instructions   
                if p_angle!=0:
                    rotateObj(obj, [x1, y1, p_angle])
                HoleList.append(obj)
        #elif v.drill[0]!=0: #circle drill hole
        elif v.drill[0] >= min_drill_size:
            #xs=p.at[0]+offset[0]+m.at[0];ys=-p.at[1]-offset[1]-m.at[1]
            xs=v.at[0];ys=-v.at[1]
            #x1=mc.end[0]+m.at[0];y1=-mc.end[1]-m.at[1]
            radius = float(v.drill[0])#/2 #sqrt((xs - x1) ** 2 + (ys - y1) ** 2)
            rx=radius;ry=radius
            #print(p.at[0],p.at[1], p.drill[0])
            [x1, y1] = rotPoint2([xs, ys], [v.at[0], -v.at[1]], m_angle)
            if holes_solid:
                obj=createHole3(x1,y1,rx,ry,"oval",totalHeight) #need to be separated instructions
            else:
                obj=createHole4(x1,y1,rx,ry,"oval") #need to be separated instructions
            #say(HoleList)
            #if p_angle!=0:
            #    rotateObj(obj, [x1, y1, p_angle])
            #rotateObj(obj, [m.at[0], m.at[1], m_angle])
            HoleList.append(obj)   
```

easyw, 2018-12-28:
Hi,
via drills are omitted by design, to avoid increasing in loading time and file size result...
Vias would add very little information related to mechanical design.

diorcety, 2018-12-28:
Some projects use vias as drill for the screws

easyw, 2018-12-28:
This is a bad design .. you would use PTH or NPTH  for anchors... 
Search for kicad pth mounting holes

easyw, 2018-12-28:
I'm closing it, feel free to re-open it if you need further assistance.

closed 2018-12-28 by easyw