easyw/kicadStepUpMod #10: Issue with make_custom ?

pyroesp, 2019-03-09:
I've had the following exception triggered when trying to load tracks:

>Running the Python command 'ksuToolsAddTracks' failed:
>Traceback (most recent call last):
>  File "C:\Users\x\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicadStepUpCMD.py", line 2471, in Activated
>    tracks.addtracks()
>  File "C:\Users\x\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\tracks.py", line 163, in addtracks
    pcb.makeCopper(holes=True, minSize=minSizeDrill)
>  File "C:\Users\x\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicad_parser.py", line 1351, in makeCopper
    thickness=thickness)
>  File "C:\Users\x\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicad_parser.py", line 1059, in makePads
>    w = make_shape(Vector(*p.size),p)
>  File "C:\Users\x\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicad_parser.py", line 188, in make_custom
>    raise RuntimeError('Cannot find polyline points in custom pad')
>
>Cannot find polyline points in custom pad

I'm not a python expert so I started by printing the shapes from make_pads() to the Report View of FreeCAD and when the program got to the first custom shape, the exception from make_custom() got triggered.  
I bypassed the custom shapes using a simple
```python
if shape <> "custom":
```
The tracks loaded fine and I could see which pads were missing.   
The missing pads came from a footprint of a custom SMD audio jack connector I made. The pads were all set to "Custom (Rect. Anchor)".  
After I changed those to "Rectangular" and removed the if-statement, there were no problems loading the tracks anymore.  
The problem of not being able to make custom pads is still there, make_custom() is just not called anymore.

The PCB can be found here : https://github.com/pyroesp/GBDSO  
I have not yet updated the PCB so if you try this with my files on the repo, you should get the error.

KiCad 5.0.2
FreeCAD 0.17
kicad StepUp version 8.1.0.8

easyw, 2019-03-09:
Hi @pyroesp 
I have tested your board and footprint.
You are using a custom rectangle anchor as a pad... you should use a round rectangle instead.
Custom anchor is used as a reference for a poly-line pad.
Switching your footprint pads to rounded rectangles will solve the issue. (I have not checked if the new pads are fitting to your original dimensions and position after the editing).
[GBDSO-rrect.zip](https://github.com/easyw/kicadStepUpMod/files/2948658/GBDSO-rrect.zip)
PS I would suggest to switch to FC 0.18 which is very near to be delivered as the next stable release and which has already many FC 0.17 fixes.

pyroesp, 2019-03-09:
Ah so it was my fault! Thanks for checking my board out. I will change the pads like you said.  

Issue can be closed.

easyw, 2019-03-09:
Thanks for your fully documented report :smile:

closed 2019-03-09 by easyw