easyw/kicadStepUpMod #11: ksu tools Add Tracks error on zone with layers

bvernoux, 2019-03-20:
I'm using latest version of kicadStepUpMod (from today trunk) with FreeCAD 0.17 rev 13522 (Git)
I'm using KiCad (5.1.0)-1, release build (but I suspect it is the same with previous version of KiCad as I doubt this parameter has changed)

I have the following error with my kicad_pcb:
```
Running the Python command 'ksuToolsAddTracks' failed:
Traceback (most recent call last):
  File "C:\Users\Ben\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicadStepUpCMD.py", line 2454, in Activated
    def GetResources(self):
  File "C:\Users\Ben\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\tracks.py", line 163, in addtracks
    pcb.makeCopper(holes=True, minSize=minSizeDrill)
  File "C:\Users\Ben\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicad.py", line 1338, in makeCopper
    thickness=thickness)
  File "C:\Users\Ben\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicad.py", line 1227, in makeZones
    if z.layer != self.layer:
  File "C:\Users\Ben\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicadStepUptools.py", line 825, in __getattr__
    del self._value[key]

7314: key "layer" not found
```
In my kicad_pcb file at line 7314 I have the line
```
  (zone (net 0) (net_name "") (layers Top In1.Cu In2.Cu Bottom) (tstamp 0) (hatch edge 0.508)
```
So yes the keyword layer does not exist as it is replaced by layers because there is multiple layers involved in this area (which is keepout zone)

It seems this is the same issue as seen in issue:
> Cannot import tracks from zone with kicad_pcb having "(layers *.Cu)" #6

Could you fix that by adding management of layers ?

bvernoux, 2019-03-20:
I have tried to fix manually my kicad_pcb 7314
from
`   (zone (net 0) (net_name "") (layers Top In1.Cu In2.Cu Bottom) (tstamp 0) (hatch edge 0.508)` 
to
`  (zone (net 0) (net_name "") (layer Top) (tstamp 0) (hatch edge 0.508)`
and the same for other zones with layers and multiple layers to layer with one layer and I confirm that fix the issue, so it will be great to manage it in kicadStepUptools.py or kicad.py

easyw, 2019-03-20:
from your error logs it seems you are using an outdated StepUp release...
in fact you have:
`Running the Python command 'ksuToolsAddTracks' failed:
Traceback (most recent call last):
....
  File "C:\Users\Ben\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicad.py", line 1227, in makeZones
    if z.layer != self.layer:
`
the actual kicad StepUp version is 8.2.0.6 should handle correctly multiple Zones...

1) Could you please check which is your release? 
2) where did you get it? 
3) have you used the FC 'Tools' 'Addons manager' menu?

bvernoux, 2019-03-23:
Thanks for the hint I confirm with latest version of FC 0.18 + fresh install of kicadStepUpMod (as I was using StepUp version is 8.2.0.6) there is no any error now

closed 2019-03-23 by bvernoux