easyw/kicadStepUpMod #12: FreeCAD 0.18 freezes after clicking "Export 3D Model to KiCad".

dervayja, 2019-03-25:
But it works just fine in 0.17

easyw, 2019-03-25:
@dervayja
Would you please post:
* your FC full version details
* your StepUp version

I use FC 0.18 regularly and I have no issues...

easyw, 2019-03-29:
I'm closing it.. feel free yo reopen in case of need.

MathieuMH, 2019-05-24:
I'm having the exact same issue. Downloaded a molex .stp positioned it, when I try to export it freecad just freezes (I left it for one hour during lunch brake). Eventually I forces closed it using task manager.

Windows 7 x64
Freecad 0.18
StepUp (installed in using the addon function in freecad, so I supose its the most recent version)
Due to the forces closure I was unable to capture the panel log. Is this kept in a file somewhere?

Any suggestions? Workaround?

easyw, 2019-05-24:
Hi @MathieuMH
would you post the full FC info (help about)
and a link to the Molex step file you are having this issue? 
Thx

mike-ady, 2019-05-26:
I have the same problem.  I've used KiCAD for years.  FreeCAD not so much.  I opened a fresh instance of FreeCAD.  I created a new document and opened the KiCAD StepUp Workbench.  Not knowing how to use KiCAD StepUp, I first created a "Part".  (Gosh, why else is there a "Part" toolbar button on the StepUp workbench toolbar?)  Then I loaded a footprint that I generated myself,  Next I loaded a .step file that I downloaded from the Phoenix Contact website.  After moving the connector into position, I clicked on "Export Model to KiCad".  Two hours later, I was still waiting.  

After restarting FreeCAD, I tried again, this time without creating any "Part".  I was able to export the .step and .wrl files without any problem.

Here is my info:

OS: Windows 8.1
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16110 (Git)
Build type: Release
Branch: (HEAD detached at upstream/releases/FreeCAD-0-18)
Hash: f7dccfaa909e5b9da26bf50c4a22ccca9bb10c40
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

I don't know how to tell the version of KiCAD StepUp, but it wasn't working when I upgraded FreeCAD to version 18.  I used the FreeCAD Addon Manager in the last couple of days to update StepUp.

mike-ady, 2019-05-26:
I'm back with the same problem but different circumstances.  This time I created my own part (a D0-214AC package).  FreeCAD was giving me grief when I attempted to create a "union" of the part, so I generated a "compound" part.  When I attempted to load the part into KiCAD StepUp, it refused to take a compound part.  So I opened the .step file in FreeCAD and I was able to delete the compound then create a union from that.  When I load the footprint, .step file, then try to export, FreeCAD hangs again.

I've attached a .zip file with the  .kicad_mod, .fcstd and .step files.  The FreeCAD module is a bit of a disaster...  I might have mentioned that I'm a total newbie.
[do-214ac.zip](https://github.com/easyw/kicadStepUpMod/files/3220433/do-214ac.zip)

easyw, 2019-05-26:
@mike-ady  
> I don't know how to tell the version of KiCAD StepUp

have you read the [cheatsheet document](https://github.com/easyw/kicadStepUpMod/blob/master/demo/kicadStepUp-cheat-sheet.pdf)? 
At page 2 you can see there is a Help button... this is giving you the StepUp version.

>  I first created a "Part". (Gosh, why else is there a "Part" toolbar button on the StepUp workbench toolbar?)

This is not clear to me? What do you mean? Have you a FC file with your model which is having this issue?

Unfortunately all the reports here  are incomplete because no one had attached a FC file showing the issue.

Please try to follow these old videos
[StepUp aligning a Step file to footprint](https://youtu.be/O6vr8QFnYGw)
[an other StepUp old video to align a model to footprint](https://youtu.be/COUdDZAhbHc)
to see if you are facing the above issue.

Note that now you can start the StepUp tools 
[image]
and then press the 'Export 3D model Aligned' button 
[image]
 
If you want you can even start a thread at KiCAD forum

easyw, 2019-05-26:
@mike-ady 
I have downloaded your zip file:
1) you are trying to align a 3D model of a smd diode to a header pins TH footprint (???)
2) you have to select your model (fusion) and then click the exporter button (at the right of the image screenshot)

I'm attaching an image of your files inside FC
[image]

easyw, 2019-05-26:
@mike-ady 
could you upload a screencast of the procedure you are facing with the hang?
I.e. using https://getsharex.com/

mike-ady, 2019-05-26:
@easyw 

I'm working on responding to your questions/requests.  This will take a while.  I have attached a .zip file with the proper footprint for the DO-214AC package.  Sorry about that.
[do-214ac.zip](https://github.com/easyw/kicadStepUpMod/files/3220537/do-214ac.zip)

easyw, 2019-05-26:
@mike-ady
thanks for your help in debugging this issue.
I'm attaching the DO214AC file with the Body fused:
[DO_214AC-fusion.FCStd.zip](https://github.com/easyw/kicadStepUpMod/files/3220949/DO_214AC-fusion.FCStd.zip)

mike-ady, 2019-05-26:
@easyw  

> have you read the [cheatsheet document](https://github.com/easyw/kicadStepUpMod/blob/master/demo/kicadStepUp-cheat-sheet.pdf)?

I have looked over the cheat sheet several times.  I apologize for saying this, but I find it incomprehensible.  It doesn't seem to tell me anything that I want to know.

Thanks for pointing out the help.  It shows "kicad StepUp version 8.3.2.3".

> > I first created a "Part". (Gosh, why else is there a "Part" toolbar button on the StepUp workbench toolbar?)
> 
> This is not clear to me? What do you mean? Have you a FC file with your model which is having this issue?

I clicked on the first button on the first toolbar that appears when I open the KiCAD StepUp Workbench.  Its tooltip reads "Create a new part and make it active".  Now that I look at it, it seems that that toolbar appears in all workbenches.

I didn't end up with a .fcstd file, because FreeCAD hung and I had to abort it.  

When I went back to recreate what happened, of course the hang didn't occur.  I don't remember exactly what I was doing when I had the problem, but after a couple of tries I found a way to reproduce it.

After reproducing the problem once, I of course had to start over after FreeCAD hung.  Then I tried saving to a .fcstd file just before I clicked on "Export 3D Model to KiCad".  On opening the saved file in another instance of FreeCAD, the hang didn't occur.  Even the action of just saving to a .fcstd file before attempting to export the .step/.wrl files seems to resolve the problem.

So I downloaded ShareX and I went through the motions of recreating the problem.  I've attached a .zip file with my .step and .kicad_mod files, and the ShareX video.

My sequence of actions was as follows:

1.  Create a new document.
2.  Select the KiCAD StepUp Workbench.
3.  Click on "Create a new part and make it active".
4.  Click on "Load a KiCad PCB FootPrint" and load the footprint file.
5.  Click on "Import 3D Step Model" and load the .step file.
6.  Select the "Part" in the Model tab of the Combo View.
7.  Click on "Export 3D Model to KiCad".

I did not take the time to orient and place the connector on the footprint since that is irrelevant to the problem.  Also, I don't recall selecting the "Part" object before clicking on "Export 3D Model to KiCad" when the problem first occurred, but I'm a newbie and anything is possible.  Is it possible that an apparently empty part is causing the export tool to loop infinitely?

Thanks for your help.
[2duppbDvw1.zip](https://github.com/easyw/kicadStepUpMod/files/3221053/2duppbDvw1.zip)

easyw, 2019-05-26:
Hi @mike-ady 
thanks for your exaustive feedback :smile_cat: 
> I clicked on the first button on the first toolbar that appears when I open the KiCAD StepUp Workbench. Its tooltip reads "Create a new part and make it active". Now that I look at it, it seems that that toolbar appears in all workbenches.

The yellow button is not related to StepUp WB, it is an internal container used by FreeCAD to handle multi objects
https://www.freecadweb.org/wiki/Std_Part/en
1) you don't need to click on 'Part' button.
2) you must select the object you want to export and then click on the "Export 3D Model to KiCad" button.
3) Note that the object you want to be exported must be a single part object (just a single shape).

I cannot reproduce this issue in a win pc, nor on linux; acting as in your screencast, FC doesn't hang... it just create an empty model...
Anyway I think I've found a workaround to avoid also this freezing...
Please update the StepUp WB and check repeating your screencast if the issue has gone.
Maurice

mike-ady, 2019-05-26:
@easyw 
Thanks.  

> FC doesn't hang... it just create an empty model...

I also saw that behavior when I deviated from the procedure shown in the video.  In any case, this is not really a problem since there is no need to create a part.  I won't do it again.

I upgraded to "kicad StepUp version 8.3.2.5".  The "part" hanging behavior is still there.

The other problem that I mentioned with the DO-214AC package does not appear to be resolved.  FreeCAD still hangs when I attempt to use the .step file that I sent earlier.  

I also created a .step from the  DO_214AC-fusion.FCstd file that you sent.  FreeCAD hangs with that too.  Perhaps I'm doing something wrong...  I selected the "Fusion_copy" object in that file and exported it as a .step file.  Am I supposed to do something else?

easyw, 2019-05-26:
@mike-ady

Please enable Report View Panel (FC View Menu, Panels, Report view) to see error messages in FC and post the screnshot of those errors.
Maurice

mike-ady, 2019-05-26:
@easyw 
I copied the text from the Report panel before I attempted to export the 3D models:

PoM not present
FC Version 018-16110
no Links support
kicad StepUp version 8.3.2.5
tolerance on vertex applied
applying Materials to Shapes
your home path is C:\Users\Mike2
export to STEP False
Tabified done !
done!
KISYS3DMOD=C:\Program Files\KiCad\share\kicad\modules\packages3d
module_3D_dir=C:\Program Files\KiCad\share\kicad\modules\packages3d/../
FC Version 018
VBO status False
FootPrint Loader C:/old_d/pcb/wcsc.pretty/DO-214AC.kicad_mod
DO_214AC_fp
thread ViewFitting
import3DModel
ReadShapeCompoundMode status False

The text in the Report panel did not change after clicking the export button.

[image]

easyw, 2019-05-27:
@mike-ady 
thanks a lot for your help...
Normally I work with a not installed FC version (to be able to use many releases on the same pc). 
Installing FC 0.18 I can finally replicate this issue ...
I'll keep you updated..

easyw, 2019-05-27:
@mike-ady 
could you please have a look at the latest commit?

I've workaround-ed the issue forcing the user to save a file before continuing to export the model.
It should work even with a clean environment (that was my real issue, because with my FC configuration files I couldn't reproduce the issue).
thanks again.

mike-ady, 2019-05-27:
@easyw
> I've workaround-ed the issue forcing the user to save a file before continuing to export the model.
That sounds like a bug in FreeCAD.

I upgraded to "kicad StepUp version 8.4.0.1".  That works much better.  It handles everything that I have to throw at it.
Thank you so much!

easyw, 2019-05-27:
@mike-ady 
thank you for your full report feedback.
I'm closing this issue then.

closed 2019-05-27 by easyw