easyw/kicadStepUpMod #13: Importing kicad_mod fails in freecad 0.19.16442 (Git)

nickoe, 2019-04-07:
FreeCAD verison info:
```
OS: Arch Linux (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.16442 (Git)
Build type: Release
Branch: master
Hash: 2935cbb71aa68f257c1d3f70b030e5beb8461b3d
Python version: 3.7.3
Qt version: 5.12.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/Denmark (en_DK)
```

ksu output:
```
kicad StepUp version 8.2.0.6
tolerance on vertex applied
KISYS3DMOD=/
module_3D_dir=//../
FC Version 019
VBO status False
FootPrint Loader /home/nickoe/stmtest/aausat.pretty/RF_Shield_TE_Connectivity_2118708-2.kicad_mod
_RF_Shield_TE_Connectivity_2118708_2__fp
Cannot compute Inventor representation for the shape of Shape.
Cannot compute Inventor representation for the shape of Shape.
Traceback (most recent call last):
  File "/home/nickoe/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 14493, in onLoadFootprint_click
    onLoadFootprint()
  File "/home/nickoe/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 7113, in onLoadFootprint
    routineDrawFootPrint(content,name)
  File "/home/nickoe/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 11468, in routineDrawFootPrint
    createSolidBBox2(obj)
  File "/home/nickoe/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 8904, in createSolidBBox2
    obj.Shape=Part.makeBox(boundBoxLX, boundBoxLY, boundBoxLZ, FreeCAD.Vector(oripl_X,oripl_Y,oripl_Z), FreeCAD.Vector(0,0,1))
ValueError: length of box too small
```
The test file: Remember to rename to kicad_mod.
[RF_Shield_TE_Connectivity_2118708-2.txt](https://github.com/easyw/kicadStepUpMod/files/3051964/RF_Shield_TE_Connectivity_2118708-2.txt)

Generated by kicad master around Version: (5.1.0-172-g3cb6dc69f), debug build

easyw, 2019-04-07:
Hi @nickoe 
it seems the kicad_mod file doesn't have any silk screen outline... this is the culprit of the import failing ...
I'm going to add an option to continue also in case of Silks missing.
Thx for reporting.
Maurice

nickoe, 2019-04-09:
@easyw ok, thank you for the prompt response.

Jut now I noticed that there is a footprint for a similar can which also do have the silk and stuff which do import just fine :)

kicad-footprints/RF_Shielding.pretty/Laird_Technologies_BMI-S-103_26.21x26.21mm.kicad_mod

I wonder if you have a script that tries to import all footprints from the libs to test the libs and importer.

easyw, 2019-04-09:
@nickoe 
I found that in the RF_Shield_TE_Connectivity_2118708-2 kicad_mod the issue is that all the values are inside a double quote...
wrong kicad_mod:
`(pad "1" smd roundrect (at 1.705 -0.5 180) (size 3.41 1) (layers "F.Cu" "F.Paste" "F.Mask") (roundrect_rratio 0.25))`
correct kicad_mod
` (pad 1 smd roundrect (at 1.705 -0.5 180) (size 3.41 1) (layers F.Cu F.Paste F.Mask) (roundrect_rratio 0.25))`
If I open the wrong kicad_mod in fp editor and simply export it, all the double quotes will disappear and the fp will be correctly loaded in FC.
I don't know if this behavior is related to your original fp which is generated with a debug build...
> Generated by kicad master around Version: (5.1.0-172-g3cb6dc69f), debug build

easyw, 2019-04-15:
Hi @nickoe 
I'm closing it because it is now solved.
Thanks for reporting it.
Maurice

closed 2019-04-15 by easyw