easyw/kicadStepUpMod #14: board import does not handle bezier curves

morganrallen, 2019-04-08:
KiCAD support's bezier curves? who knew!

Well, it appears when importing from DXF, KiCAD will create bezier curves but StepUp doesn't seem to like or handle them.

[image]

Note the two handle points on the selected curve.

[image]

All source files can be found here, https://gitlab.com/morganrallen/kickbadge/commit/c72e021e5a41f7ae2860fb3e7477c6b5a273467b

easyw, 2019-04-08:
@morganrallen thanks for let me know about the new bspline support for edge-cut from kicad...
I'm going to see f I can support it...
BTW the correct workflow would be to start your design in FC and not importing a dxf in kicad ...
1) design your edge in FC (or import your dxf in FC) 
2) push the edge to kicad using stepup
3) continue your job in kicad
NB please remember to add reference in kicad like i.e. a grid origin to be use as a mutual reference between kicad and freecad.

easyw, 2019-04-12:
Hi @morganrallen 
please have a try with the new update...
StepUp now can import also splines and push back the edge to KiCAD if required.
Still is a first implementation that can be improved.

easyw, 2019-04-15:
[image]

morganrallen, 2019-04-16:
Hey Maurice, thanks a lot, it's looking great now.

closed 2019-04-16 by morganrallen