easyw/kicadStepUpMod #15: Can't open footprin in FreeCAD 0.18.1

jneiva08, 2019-04-11:
kicadStepUp: ksuWB.Activated(v 7.8.6)

Freecad Version:
OS: Linux Mint 19.1
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.1.
Build type: Release
Python version: 3.6.7
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

Since i updated the freecad to release mentioned, the kicadStepUp is loaded but when try open an footprint or pcb gives the following error:

Running the Python command 'ksuToolsOpenBoard' failed:
Traceback (most recent call last):
  File "/home/jneiva/.FreeCAD/Mod/kicadStepUpMod/kicadStepUpCMD.py", line 248, in Activated
    import kicadStepUptools
  File "/home/jneiva/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 1565, in <module>
    FC_majorV,FC_minorV,FC_git_Nbr=getFCversion()
  File "/home/jneiva/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 1558, in getFCversion
    FC_minorV=int(FreeCAD.Version()[1])

invalid literal for int() with base 10: '18.1'

easyw, 2019-04-11:
@jneiva08 
FC 0.18.1 has a different way to implement its release version info.
Anyway, it should be fine now.
Please update StepUp @8.2.0.9 through FC Tools Addons manager.

closed 2019-04-11 by easyw

jneiva08, 2019-04-11:
Thanks, it's working