easyw/kicadStepUpMod #16: Allow importing board in addition to load

morganrallen, 2019-04-18:
As it works now, when using the load board feature, a new FreeCAD project is created and the board is recreated in there. When working with multiple boards that join together, this requires creating a new project for each part, copying the parts out of the new project and pasting into my working project.

It would be great to have an option to insert the board into the current project rather than creating a new one. Personally, this makes more sense as the default behavior to me.

easyw, 2020-01-08:
@morganrallen 
from now https://github.com/easyw/kicadStepUpMod/pull/34 it is possible to 'Import' a KiCAD board inside an existing FreeCAD file, beside the standard behavior to 'Open' a KiCAD board, creating a new FreeCAD file.

closed 2020-01-08 by easyw

morganrallen, 2020-01-08:
Super cool! I'll check this out with my current project.

ifohancroft, 2021-08-28:
How do you import the board into the current file? The only think I could find that does that is the 'Pull Sketch' option, but it doesn't pull the elements, so then when I try to use 'Pull 3D models placement from PCB' it doesn't let me and gives me this message 'select only 3D model(s) to be updated/pulled from kicad board!
a Time Stamp is required!'