easyw/kicadStepUpMod #17: Cannot find polyline points in custom pad

matthuszagh, 2019-05-21:
I'm getting the error shown in the issue title when trying add tracks to a loaded board (attached here):
[patch_1x1x5.zip](https://github.com/easyw/kicadStepUpMod/files/3202125/patch_1x1x5.zip)

I'm using the current Git master version of this program and Freecad 0.18.

This seems to be closely related to [this issue](https://github.com/easyw/kicadStepUpMod/issues/10). I imagine I could make a similar fix, however, this does seem to be a valid pcb in pcbnew so I'm reluctant to do that. Is there a reason my pcb is wrong?

On a somewhat related note, I noticed when looking through the implementation that you parse the S-expressions yourself rather than using Pcbnew's python API. Have you considered using the API?

Thanks!

easyw, 2019-05-22:
Hi @matthuszagh 
StepUp is not covering all kicad pad models...
Particularly not full covered the Geometry ones...
I also noted that I'm not covering SMD pads on back side (B.Cu).... this last could be improved quite sooner.

Your model could be designed as in the attached FreeCAD file and in that case I can re-import the model in FreeCAD as '.kicad_mod' to check its geometry.
[RF-antenna-files-v3.zip](https://github.com/easyw/kicadStepUpMod/files/3207605/RF-antenna-files-v3.zip)
[image]

 
-
> On a somewhat related note, I noticed when looking through the implementation that you parse the S-expressions yourself rather than using Pcbnew's python API. Have you considered using the API?

KiCAD APIs are quite unstable and depending on kicad release version... moreover they are not accessible from FC internal environment.
BTW, even with the kicad APIs, the need to convert a pad syntax to a FreeCAD geometry needs to be done.

matthuszagh, 2019-05-22:
Thanks for the response @easyw, that all makes sense. And thanks for posting the correct model!

closed 2019-12-23 by easyw