easyw/kicadStepUpMod #19: [Wish] Two different buttons for inc/dec rotations and translations

f6fvy, 2019-06-01:
kicad StepUp version 8.4.0.2

In the tools window, having two different buttons for increasing or decreasing rotations and translations would be much more convenient to use, instead of editing the increment (positive or negative).

cdwijs, 2019-10-14:
Thanks f6fvy, I was about to write a bugreport for the same thing.

cdwijs, 2019-10-14:
I don't know howto edit the plugin and test it in Freecad, but the needed changes are in kicadStepUptools.py in class Ui_DockWidget(object): and in routineT_XYZ(axe,v):

Where does FreeCAD store this plugin?

easyw, 2019-10-14:
Hi @f6fvy @cdwijs 
I'm sorry but I cannot add buttons to the already crowded StepUp Gui.
But there is a viable solution for this...
Install my [Manipulator WB](https://www.freecadweb.org/wiki/Manipulator_Workbench) and use the Mover tool. This tool has been developed later on StepUp Gui and it is more powerful (can work with Body and Part containers, move/rotate based on edge directions etc.).
Moreover it has buttons  😄 
[image]

closed 2019-11-16 by easyw