easyw/kicadStepUpMod #21: [Wish]

leoheck, 2019-07-28:
Hello!

First of all, this plugin is awesome, thank you for the efforts to make it good.

I saw that many plugins, when added to Freecad are inserted on the Workbench list in an ordered manner. This one, however, is being added to the end of the list. Is it possible to update it to do the same? It is faster to select the workbench when the list is ordered.

Examples of plugins that are being inserter in an ordered way
- Fasteners
- Part-o-magic

leoheck, 2019-07-28:
I think this should be done like this: https://github.com/shaise/FreeCAD_FastenersWB/blob/b3cd9b48af8ba448451f5a3f43b8096c2ab6e283/InitGui.py#L64

I saw in your InitGui.py that you are following the A2Plus as a reference but it adds the workbench at the end which is not good. I reported the same there issue

easyw, 2019-07-29:
Hi @leoheck 
it seems you can configure the WB order just modifying it in your FC preferences (Tools, Customize, Workbenches)
-
[image]

leoheck, 2019-07-29:
Hi @easyw, cool thanks for the tip. So, why not to add it in alphabetic order by default?

easyw, 2019-07-30:
Hi @leoheck 
I don't know which the code could address it, using python, so ATM I have to rely on FC GUI  😸

leoheck, 2019-07-30:
Cool, if I find something I will make a PR on that. @easyw thanks for the answer.

easyw, 2019-08-12:
Hi @leoheck 
it should be solved with this PR https://github.com/easyw/kicadStepUpMod/commit/b34810860e1bfdf74c860823164a834370540a8e
please have a try :smile_cat:

leoheck, 2019-08-12:
@easyw that is awesome.

However, I give it a try here but I was not able to see any difference.
The version I am using is the following. I see the `solved ordering WBs` is the previous commit, but the order is still the same.

This is my Freecad version
```
OS: Ubuntu 19.04
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.3.
Build type: Release
Python version: 3.7.3
Qt version: 5.12.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United States (en_US)
```

```
commit 78e7b8775438d7d0062164e57b02faf1a7f2b5e9 (HEAD -> master, origin/master, origin/HEAD)
Author: easyw <3032347+easyw@users.noreply.github.com>
Date:   Mon Aug 12 18:57:40 2019 +0200

    new tool to convert edges to sketch

commit b34810860e1bfdf74c860823164a834370540a8e
Author: easyw <3032347+easyw@users.noreply.github.com>
Date:   Mon Aug 12 15:13:01 2019 +0200

    solved ordering WBs
```
I also tried to remove and reinstall it again but nothing has changed.
Am I doing something wrong?
[image]

leoheck, 2019-08-12:
Oh, it is a PR yet!!

easyw, 2019-08-12:
Humm this is weird..
It works for me on:

OS: Linux Mint 19.2
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.3. & 0.19 latest
Build type: Release
Python version: 3.6.8
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0

[image]

easyw, 2019-08-12:
@leoheck
would you please check with a clean profile?
(you can save the old ".FreeCAD" profile and test it with an empty profile).

leoheck, 2019-08-13:
Testing here, at home now, with a different computer and it is working just fine. Tomorrow I will test on the same computer I did before (removing the old profile as you said.)

leoheck, 2019-08-13:
After removing the old .FreeCad folder it is working @easyw

Ah, thank you so much for renaming `kicadSketchup` to `KicadSketchup` now it looks professional.

You can also close this issue again.

Even better changing it from `KicadStepUp` to `Kicad StepUp`. 
Also, the menus `ksu Tools` and `ksu Push Pull` could be a single one 
`Kicad StepUp > Tools > ... `
`Kicad StepUp > Push Pull > ...`

easyw, 2019-08-13:
@leoheck 
thanks for your testing & feedback.
I'm closing as solved :smile:

closed 2019-08-13 by easyw