easyw/kicadStepUpMod #22: [issue/question] refraction of material part of Material or set in script?

offerman, 2019-08-20:
Added some new materials to the script, including transparent glass (for radio/VFD tubes):
```
ambientIntensity 0.00
diffuseColor 0.588235 0.670588 0.729412
specularColor 0.90 0.90 0.90
emissiveColor 0.0 0.0 0.0
shininess 0.85
transparency 0.80
```
The result looks fine in KiCad's 3D view, but when the model is rendered the glass material looks more like plastic, i.e. a lot of deformation because the refraction is too high.
Is this an issue with my Material definition, or is this part of the script?

offerman, 2019-08-20:
[image]
[image]

easyw, 2019-08-20:
Hi @offerman
the script should already handle 3 glass materials...
Anyway the refraction value that you see when you enable raytracing is coded inside the KiCad 3d-viewer.
Unfortunately this is not available as settings.
You may rise an issue at KiCad bugs launchpad.

offerman, 2019-08-20:
Thanks @easyw 
There are actually 5 glass materials already! :) but none of them fully transparent. I checked out the grey glass Material: transparent but milky. So I'll submit a pull request for the transparent glass and the black glass (for the getter part) when i got this rendering issue sorted out.
Apart from the refraction apparently being hardcoded in the viewer, from [this post](https://forum.kicad.info/t/3d-raytracing-with-transparent-enclosures/8994) I understand that the renderer processes hollow objects as massive objects.
So per your suggestion I reported to the KiCad bug list [here](https://bugs.launchpad.net/kicad/+bug/1840848).
I'll follow up on this one when I know more.

closed 2019-12-23 by easyw