easyw/kicadStepUpMod #23: Parsing of module names not always correct

ekelectronics, 2019-08-21:
Caveat - Please interpret anything presented as fact as "as far as I can tell" 

On loading a .kicad_pcb file into freeCad, the parsing of file names for 3D models associated with footprints doesn't always yield sensible results, or results that match KiCad's behaviour

In the attached file there's a board with 6 LEDs, all with variations on the same 3d model, either .wrl or .stp. Note that isn't .step
In the KiCad 3D viewer, all 6 diodes appear correctly, if you load the board into freeCad only 2/6 appear. 

The relative file paths beginning ./ load into step up, but both use the .stp model. (Removing the .stp causes neither to show)

For other files, or for relative files that can't be found, and absolute file paths, the loader seems to: 
strip any leading / 
strip the extension
prepend $KIS3DMOD 
add extension .step

The dialogue only informs the user on the modified try. If I set the file path in KiCad to 
./relative/FileThat/DoesntExist.stp or ./relative/FileThat/DoesntExist.wrl
returns
...missing module(s)
Drive:/files/to/$KIS3DMOD/AbosultePath/toProject/./relative/FileThat/DoesntExist.step cannot be found

This is confusing, as it makes the user think that the environment isn't set up properly, when in fact the file isn't present, so the loader checks elsewhere, and the subtle extension change is easy to miss. 

or using 
Drive:/Absolute/FileThat/DoesExist.stp
returns 
...missing module(s)
Drive:/files/to/$KIS3DMOD/Drive:/Absolute/FileThat/DoesExist.step 
and I'm not 100% sure how to force an absolute file path yet. 

You can play around with the files and get various nonsensical paths. With this board, I never seem to get more than two "missing models" even though I'm using 6 different file extensions/paths. This would suggest to me that the importer is marking them as bad and skipping future references, even though the file extensions, and therefore files themselves, are different. This could mean an erroneous .wrl causes valid .steps not to be imported.

I've also seen when using $ALT3DMOD or $KIS3DMOD the extension isn't preserved, hence .stp files will not load from these locations. Resulting in the message 
...missing module(s)
Drive:/path/to/$ALT3DMOD/fileThatDoesExist.step
when the file does exist, but as .stp or .wrl

If you build a kicad board with a SOT-23 footprint as it comes from the library, the footprint 3d model is
${KISYS3DMOD}/TO_SOT_Packages_SMD.3dshapes/SOT-23.wrl
but also you'll find in that folder
${KISYS3DMOD}/TO_SOT_Packages_SMD.3dshapes/SOT-23.step
exists and if you remove/rename it the package won't appear in freeCad Even if the .wrl is unchanged.

This is particularly confusing, as some Kicad parts have both, and some don't resulting in some parts loading just fine, with others not and no obvious difference. For example 
Socket_Strip_Straight_1x14_Pitch2.54mm doesn't have the appropriate .step so doesn't load

I'd suggest the behaviour should be
- any file path options that work in KiCad should worth with KicadStepup
-- hence if the file begins with a / or // or .// and so on, the file should be treated as relative
-- if there is an environment variable is present that should be prepended
-- else it should be treated as absolute as passed as is
- the exact extension should be preserved (ie .step / .stp   .wrl / .vrml ) in all events. 

I'm not actually sure if step Up is able to import .wrl files in this case (I know there are some options in exporting boards), if it's not designed to work with .wrl then it should inform the user that it can't import them. I don't think loading a corresponding .step file with the same name (if present) is a good idea. Giving an "I can't import .wrl" message gives the user the choice to change to a suitable file, importing a different file to the one the user inputs could result in old/different/bad models being used.

[StepOrStp.zip](https://github.com/easyw/kicadStepUpMod/files/3526130/StepOrStp.zip)

easyw, 2019-08-21:
Hi @ekelectronics 
thanks for your reach feedback.
> I'm not actually sure if step Up is able to import .wrl files in this case (I know there are some options in exporting boards), if it's not designed to work with .wrl then it should inform the user [...]

Please consider that, as in the [kicad library convention](http://kicad-pcb.org/libraries/klc/#_general_library_guidelines):
> Filetype must be .wrl (KiCad replaces it with .step automatically for the step export.)

StepUp will replace .wrl to .stp or .step and will try to load those.
Consider that for mechanical purposes, wrl are almost useless.. 
Here some tips why .wrl is used in kicad libraries, but .step are used for MCAD:
[what-is-the-difference-between-step-and-wrl-3d-models](https://forum.kicad.info/t/what-is-the-difference-between-step-and-wrl-3d-models/17346)
> Exporting to step either via the internal step exporter or via the freecad extension stepup will automatically replace the wrl file endings of the 3d path with step file endings.
Meaning you can add wrl models for the internal viewer and still be able to export step as long as you have both files stored under the same name and scaled correctly. (KiCad reads 1 wrl unit as 1 inch. Step stores its base unit explicitly. Stepup already exports correctly scaled wrl and step model pairs)

StepUp is using two different variables as prefixes for 3d model research path, but you need to configure those inside FreeCAD StepUp preferences.
Please have a look at [StepUp cheatsheet](https://github.com/easyw/kicadStepUpMod/blob/master/demo/kicadStepUp-cheat-sheet.pdf) at  pg.10 you can see where the configuration prefixes have to be configured.
Saying that, StepUp should honor the following paths:
1) absolute path
2) path relative to main 3D folder prefix
3) path relative to secondary 3D folder prefix
4) relative path to kicad_pcb file

And should import the .stp or .step (or .igs or .iges) file or a file with the same name of a wrl file, but with an extension replaced with one of the previous ones.

Please feel free to add more requests if you need.
Maurice

ekelectronics, 2019-08-22:
Hi Maurice,

thanks for a quick response!

Yes so I see step up is doing the same as KiCad, I've done a Kicad export and the models that have equivalent .step files get exported, files that only exist as .wrl don't make the cut. While I wouldn't think much value in adding in .wrl compatibility, consider informing the user that the extension they supply will be replaced. Cheat sheet section 5) Tips
"... use STEP or IGES or VMRL or mixed type ..."
reads to me that I should expect the converter to import .wrl, when it won't.  

Currently on my machine
$KISYS3DMOD is C:/Program Files/KiCad/share/kicad/modules/packages3d/
and the project location for StepOStp.kicad_pcb is
D:/Google Drive/AGI/Projects/Non Specific/Electronics/StepOrStp
which then has a sub folder 
/New Folder/

Consider the following message :
[image] 

which is generated by trying to import the file StepOrStp.kicad_pcb into StepUp. The importer has neither honoured the file extension (changing both .wrl and .stp to .step) and is looking in a bizarre location.
For your 4 notes

> Saying that, StepUp should honor the following paths:
>
>absolute path 
>path relative to main 3D folder prefix 
>path relative to secondary 3D folder prefix 
>relative path to kicad_pcb file (with './' or '${KIPRJMOD}/' prefix) 

For all of these it does work but
- only if a .wrl file can be renamed .step
- a relative file must begin ./ even to file is in the same folder
- an absolute path must be .step

if any of these rules are false, the importer gives the user the message where the path is appended to $KISYS3DMOD. 

I think I can see this is coming from kicadStepUptools.py line 6143
` if test is -1:
      missing_models += make_string(models3D_prefix)+make_string(step_module)+'\r\n' #matched    `
this tells the user that the module has looked in $KISYS3DMOD/step_module.step
when in fact the program has checked the file as
 .step .stp and .iges
and as a relative, absolute, $KISYS3MOD/ and $ALT3DMOD/ path 

as well, at line 5444 you generate all 4 possible extensions, but you've stripped any closing " from the paths. Hence kicad that handles absolute paths as
"Drive:/Abs/Path/Model.stp"
produces step_model2 =
"Drive:/Abs/Path/Model.stp

you strip this beginning " from step_module, but not step module 2-4. Adding
`step_module2=step_module2.replace(u'"', u'')  # name with spaces`
at line 5452 fixes the import on a stp for me

if this makes sense I'd be happy to generate a pull request

easyw, 2019-08-22:
Hi @ekelectronics 
I've improved both path searching and error messaging...
would you please update and test the latest commit?
Maurice

ekelectronics, 2019-08-26:
Hi @easyw 

this is pretty much there, though being the nit-pick I am

For the relative path, 2/3 options are accomodated
relativePath/Tofile.ext
./relativePath/Tofile.ext
and not
/relativePath/toFile.ext

I'd feel bad for raising it, but it will still be quite unclear to the user why this isn't working. The error message that shows all three file locations I think is grand as it allows the user to easily see if their paths aren't set up properly. I can see that in the error message, the last_pcb_dir is printed without a slash at the end, but the other two do, and I wonder if that's indicative of the problem.

I'd also suggest, and it's really more a matter of my opinion here, that the list of missing modules be shown without extention, to force the user to read the message and understand that these files have been looked for under those for pre-set extensions, not what was given. You can populate this by keeping a copy of the unextentioned name at line 5495, and adding this to missing from 6195.

Ed

ekelectronics, 2020-02-17:
Hi Maurice,

another really finicky one on this subject, I'm commenting more in case anyone finds this thread than expecting you to accommodate it. 
Kicad accepts the use of both brackets ( ) and braces { } in the environment variables - ie both '$(ALT3DMOD)' and '${ALT3DMOD}' are valid and will load footprints in KiCad, but step up only looks for braces!

Ed

easyw, 2020-02-18:
@ekelectronics 
would you please upload a minimal pcbnew file which has your path issues?

ekelectronics, 2020-02-18:
[pathIssue .kicad_pcb.txt](https://github.com/easyw/kicadStepUpMod/files/4219790/pathIssue.kicad_pcb.txt)
[image]

easyw, 2020-02-18:
with [latest update](https://github.com/easyw/kicadStepUpMod/commit/126e0792f0a32a7081ebb60abf62349e7294a901) this issue should be solved ... but just for information, in which way you get this two way to add the variable?
 
Moreover, I was referring to your previous posted issue...
> as well, at line 5444 you generate all 4 possible extensions, but you've stripped any closing " from the paths. Hence kicad that handles absolute paths as
> "Drive:/Abs/Path/Model.stp"
> produces [...]

ekelectronics, 2020-02-18:
When setting 3D models for footprints in KiCad, you can just type your path in. I've only stumbled across this issue because KiCad opens to its default 3D model location in a pretty naff explorer, and it's far quicker to type in the variable than it is to navigate to the folder. 

I'm not 100% sure what you're after, as that issue you've quoted is also fixed. 
The only outstanding issue is that of relative file paths in the format
/relative/path.ext
work with kicad, but are not imported to freeCad. You can replicate the issue by creating a subfolder, putting a model into it, and setting it as a 3D model to a footprint using the above syntax.

easyw, 2020-02-18:
@ekelectronics 
> When setting 3D models for footprints in KiCad, you can just type your path in. I've only stumbled across this issue because KiCad opens to its default 3D model location in a pretty naff explorer, and it's far quicker to type in the variable than it is to navigate to the folder.

this '$(' vs '${' is a glitch in KiCAD standardization of variables... from k5.1 on '${' is the way to go .. anyway StepUp can now handle also the old way mode...

> The only outstanding issue is that of relative file paths in the format
> /relative/path.ext
> work with kicad, but are not imported to freeCad. You can replicate the issue by creating a subfolder, putting a model into it, and setting it as a 3D model to a footprint using the above syntax.

'/relative/path.ext' is not a correct way to point to a 'folderpath' + plus a 'relative/path.ext'
The right way is to write 'folderpath' + 'relative/path.ext'
The '/relative/path.ext' has to be intended as an absolute path, particularly on Linux...

Did you added this path manually too?

closed 2020-02-19 by easyw