easyw/kicadStepUpMod #24: Antenna Footprint

leoheck, 2019-09-02:
Hi, I am importing the board tracks, but I have some footprints with some tracks F.cu and B.cu tracks. For example, I have a trace PCB antenna which is not imported with the tracks with the Add Tracks command. Do you have any alternative to import tracks in footprints?

easyw, 2019-09-02:
Hi,
footprint with tracks are not allowed by kicad... those can be done as a hack.
I would suggest to use a poly-line pad to design your antennas...
Please have a look at the files inside StepUp demo menu.
https://github.com/easyw/kicadStepUpMod/blob/master/demo/footprint-RF-antenna.FCStd
https://github.com/easyw/kicadStepUpMod/blob/master/demo/footprint-RF-antenna-w-solder-Mask.FCStd
[image]

morganrallen, 2019-09-02:
FWIW, I believe future versions of KiCAD will support non-pad copper in footprints. I could be mistaken but I recall hearing something along these lines at KiCON.

easyw, 2019-09-02:
Antenna normally are designed from a mechanical sw ... typically you can find antenna models in DXF format... the DXF format is also used by most RF simulation sw...
then the best option is to convert the DXF to a sketch and then use it to export the model as a poly-line pad for kicad.

leoheck, 2019-09-02:
Thanks for the workaround @easyw 

The antenna I want to export was made by my in Kicad, and I was not able to export .dxf files from the Library Editor.

I actually filed an issue today. Regarding exporting a .dxf from Library Editor.
https://bugs.launchpad.net/kicad/+bug/1842348

easyw, 2019-09-03:
You can simply export a board with your antenna fp as DXF, and then extract your data from the resulting DXF.

easyw, 2019-09-03:
FYI here some Ti antenna DXF shapes
[CC_2D00_Antenna_2D00_DK2.dxf](https://e2e.ti.com/cfs-file/__key/communityserver-discussions-components-files/538/CC_2D00_Antenna_2D00_DK2.dxf)

luzpaz, 2022-04-30:
> I actually filed an issue today. Regarding exporting a .dxf from Library Editor. https://bugs.launchpad.net/kicad/+bug/1842348

JFYI @leoheck, your upstream ticket has moved to https://gitlab.com/kicad/code/kicad/-/issues/2493

closed 2022-07-03 by easyw