easyw/kicadStepUpMod #25: Imported STEP in Fusion 360 has no track/silk color on bottom layer

TinusScheppers, 2019-10-06:
The STEP file is created by exporting a PCB+Tracks+Silk using FreeCAD.

When importing this step model in Fusion it seems that the color reverts to (default?) copper for all items except the PCB.

However when importing the STEP file back in FreeCAD everything is in the correct color.

Is there something that can be done to correct/change this behaviour?

Example: 
Front in FreeCAD
[image]

Front in Fusion
[image]
=> Behaviour as expected

Back in FreeCAD:
[image]

Back in Fusion:
[image]
=> Behaviour not as expected

Please find the STEP file attached:
[Front.zip](https://github.com/easyw/kicadStepUpMod/files/3694656/Front.zip)

TinusScheppers, 2019-10-06:
In addition: I just noticed that when creating a mirrored version (in Fusion), the original(!) gets the correct colors (the mirrored version still has the wrong colors)...

easyw, 2019-10-07:
Hi @TinusScheppers 
this seems an issue in Fusion360 STEP importing function.
You can try to open the file with [CAD Assistant](https://www.opencascade.com/content/cad-assistant) or with [DesignSparkMechanichal](https://www.rs-online.com/designspark/mechanical-download-and-installation) and there are no issue.

TinusScheppers, 2019-10-07:
> Hi @TinusScheppers
> this seems an issue in Fusion360 STEP importing function.
> You can try to open the file with [CAD Assistant](https://www.opencascade.com/content/cad-assistant) or with [...]

Indeed, definitely an issue in Fusion given the above workaround.
Anyway, I was hoping maybe you had an idea what the cause might have been, but its more of a problem for the Fusion 360 developers.
Maybe I'll post it over there.

Keep up the good work, this is a great tool!

closed 2019-10-07 by TinusScheppers

jacobtohahn, 2020-08-01:
@Devenda, did you ever find a solution for this issue anywhere? I have the exact same problem and I'm not sure how or if I can resolve it.

Devenda, 2020-08-01:
@jacobtohahn, The best I could find was a workaround.

It was possible to create a mirror of the offending piece, strangely enough this made the original piece appear as expected.
After this the (new) mirrored piece can be removed. (You will see that the mirrored piece again has the color).

[image]

Good luck!

jacobtohahn, 2020-08-01:
Interesting. Glad the workaround is that simple.