easyw/kicadStepUpMod #30: Add Tracks operation failed

AlexUg, 2019-12-02:
Hi!
Trying to perform operation 'Add Tracks' after importing 'PCB Boards and Parts' produces error: 
```
Running the Python command 'ksuToolsAddTracks' failed:
Traceback (most recent call last):
  File "/home/alexandr/.FreeCAD/Mod/kicadStepUpMod/kicadStepUpCMD.py", line 3047, in Activated
    import tracks
  File "/home/alexandr/.FreeCAD/Mod/kicadStepUpMod/tracks.py", line 48, in <module>
    if FC_git_Nbr >= int(FC_export_min_version):

name 'FC_export_min_version' is not defined
```
After adding lines
```
global FC_export_min_version
FC_export_min_version="11670"
```
to 'tracks.py' file operation 'Add Tracks' begin work without errors. At first glance all tracks and vias were imported correctly.
But I think this bugfix isn't right...

PS: used revision is '3141f0715eba54ceca9e67406eea2b14270235d2'

easyw, 2019-12-02:
@AlexUg 
I cannot replicate your issue...
Could you please add your FreeCAD full file info? (from FC Help Menu, About FC, copy to clipboard)
Have you a sample kicad_pcb file to share to test it?
Thanks

AlexUg, 2019-12-03:
FreeCAD info:
```
OS: KDE neon User Edition 5.17
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13541 (Git) AppImage
Build type: None
Branch: releases/FreeCAD-0-17
Hash: 9948ee4f1570df9216862a79705afb367b2c6ffb
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Russian/RussianFederation (ru_RU)
```
Sample KiCAD project:
[Sandbox.zip](https://github.com/easyw/kicadStepUpMod/files/3914575/Sandbox.zip)
But the error appears before the PCB is processed so the content of the PCB is irrelevant. And bug is gone when 'FC_export_min_version' is defined in file 'tracks.py'.

easyw, 2019-12-03:
Ok, I see you are using an old release... FC 0.17 ...
please have a try with FC 0.18.4 which is the actual stable release.

AlexUg, 2019-12-03:
In FC 0.18 the error does not occur.

It would be nice to update the requirements section in the project description:
> FreeCAD 0.15 4671 or 0.16 >= 6712 or 0.17 >= 11707 or 0.18 or 0.19

closed 2019-12-03 by AlexUg