easyw/kicadStepUpMod #31: Provide option to disable the sidebar popup

iromero91, 2019-12-13:
This panel pops up every time I open a board and when I do other operations , sometimes whenever the workbench is loaded.

[image]

most of the tools are repeated in the other toolbars and the manipulation tools are only really useful when working with STEP models for footprints. It takes a lot of space on the screen and I would like for it to stay closed until I need it. Would it be possible to add an option to do that?

easyw, 2019-12-13:
@iromero91 
ATM you can attach the Dialog to the left side ... it will be placed under the Combo  
But I'm working on an option to hide the main Dialog that was present since the beginning of the development...

easyw, 2019-12-15:
@iromero91 
with latest commit https://github.com/easyw/kicadStepUpMod/commit/299c1cb4904b90cad3d3677ea32f56aacd7b2e12 the main dialog appears only on request (clicking the button on the WB toolbar) or opening a footprint in FC.

closed 2019-12-15 by easyw