easyw/kicadStepUpMod #33: Can't push model positions to board file

Shackmeister, 2020-01-06:
When trying to push the component positions, I get this error eventhough the import is set to "Grid origin"
[image]

easyw, 2020-01-06:
> When trying to push the component positions, I get this error eventhough the import is set to "Grid origin"

@Shackmeister 
But your KiCAD pcbnew file has 'Grid Origin' set accordingly?
The message say the board must have 'Grid Origin' assigned (and not only the FC import preference set).
Could you please check if your KiCAD pcbnew file has Grid Oring set? 
Then you would need to reload the Board inside FC and play with Push&Pull placement...

Shackmeister, 2020-01-06:
The Grid has the Origin set to 0,0 (Using the tool above the caliper) Should this be fine?

Shackmeister, 2020-01-06:
OK, so apparently 0,0 is not allowed, Any specific reason for this?

easyw, 2020-01-06:
@Shackmeister 
I meant to set the Grid Origin in KiCAD pcbnew...
Have you done it?
[image]
from KiCAD doc:
[image]

Shackmeister, 2020-01-06:
Everything works once I set it to anything other than 0,0.
Everything is however made from the 0,0 grid origin

easyw, 2020-01-06:
@Shackmeister 
I see the issue now...
KiCad places a default Grid Origin in 0,0 at the top left corner in pcbnew.
When you place a 'Grid Origin' in your design, the file is modified ONLY if Grid Origin is placed in a different location from the default one.
StepUp is looking for the internal value of Grid Origin and if it is not present, then it will suggest to place one.
The Grid Origin parameter is set inside the setup group in the pcbnew file as i.e.:
`    (grid_origin 149.606 97.282) `
Since I never used a board with a reference placed at top left corner, I never noticed that Grid Origin value is written inside KiCAD only if not set on 0,0.
I can patch this weird KiCad behavior... 
But in general I would suggest you to place your reference i.e. at a circular hole of your board... then your Sketch will have all the objects distances referred to a convenient Grid Origin reference.

easyw, 2020-01-08:
@Shackmeister 
would you please check the latest dev release?
https://github.com/easyw/kicadStepUpMod/archive/uid.zip
This release should fix your issue, among others and add new features

Shackmeister, 2020-01-13:
Hi Maui
> When you place a 'Grid Origin' in your design, the file is modified ONLY if Grid Origin is placed in a different location from the default one.

I think this is coming practice for S-expression (only include stuff if != 0)

> But in general I would suggest you to place your reference i.e. at a circular hole of your board... then your Sketch will have all the objects distances referred to a convenient Grid Origin reference.

I will consider this moving forward, but currently my board are referenced to 0.0. This makes the boards show up in the correct position of my MCAD model

easyw, 2020-01-13:
> but currently my board are referenced to 0.0. This makes the boards show up in the correct position of my MCAD model

you can simply add a grid reference to the point you need to be your 0,0 in MCAD, then the board will appear as desired in your MCAD and at the right placement in KiCAD

closed 2020-02-19 by easyw

Anool, 2020-04-08:
I get this error even after I've set Drill Offset AND Aux. Origin to a non 0,0 value (lower left corner of my PCB. Tried in various FreeCAD versions - AppImages for v0.17, v0.18, v0.19 as well as installed 0.18.4 (details below)

OS: Ubuntu 18.04.4 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4.
Build type: Release
Python version: 3.6.8
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/India (en_IN)
[image]
[image]

easyw, 2020-04-08:
> I get this error even after I've set Drill Offset AND Aux. Origin to a non 0,0 value (lower left corner of my PCB. Tried in various FreeCAD versions - AppImages for v0.17, v0.18, v0.19 as well as installed 0.18.4 (details below)

@Anool, could you please attach the zipped file or link to a github repo?

Anool, 2020-04-08:
Here goes.
[project_snap9V_kicad.zip](https://github.com/easyw/kicadStepUpMod/files/4449191/project_snap9V_kicad.zip)

I'm trying to demo KSU advanced features on the next KiCad/FreeCAD AMA on Saturday.

easyw, 2020-04-08:
@Anool, for me it is working...
could you please check your preferences, as in the attached image?
[image]

Anool, 2020-04-08:
Awesome ! That works. Maybe it needs a note in the Cheat Sheet.
I'll demo this (part position push/pull) and Edge push/pull on the next Meetup.

easyw, 2020-04-09:
> That works. Maybe it needs a note in the Cheat Sheet.

I'll fix the error message to point the user to the preferences menu asap

> I'll demo this (part position push/pull) and Edge push/pull on the next Meetup.

Thanks a lot! :smile: Looking forward to see your video on youtube.

easyw, 2020-04-09:
@Anool 
I've pushed a fix for the error messages https://github.com/easyw/kicadStepUpMod/commit/827456ca9ff4d9c2e64a008e933b9a24a24a8026