easyw/kicadStepUpMod #37: Shape offset doesn't work in THT footprints

EeliK, 2020-01-17:
I have THT footprints with shape offset, StepUp doesn't apply the offset to copper when tracks (and therefore footprints) are loaded into the model.

kicad StepUp version 9.5.2.4

[image]

[image]

[image]

easyw, 2020-01-17:
@EeliK 
StepUp footprint loader should respect the offset.
StepUp Tracks importer it is not ATM supporting it.

EeliK, 2020-01-17:
I clarifed the title to reflect the problem. Indeed, I imported tracks to the 3D board which imports pads, too. Luckily the board is still mechanically correct.

easyw, 2020-01-17:
> I clarifed the title to reflect the problem.

@EeliK the title is now even more misleading. 
Would you change it to something like:
'Loading tracks will not respect the pad's offset'
because it seems StepUp will not respect footprints, but this is not true. 
When importing a single footprint the pads will respect the offset.
Only when importing tracks the issue is present. Consider that tracks are loaded only for aesthetic reasons.  
The footprint are instead loaded for mechanical alignment.

EeliK, 2020-01-17:
You can't always predict what users want and can do with your program :)

This isn't for esthetic reasons but for checking how the parts are connected together - not only mechanically but also how the copper continues (through connector holes/fingers) from one board to another.  Eye candy is nice but it can be also surprisingly useful. That's why I'm thankful for both KiCad's integrated 3D view and your StepUp's features.

easyw, 2020-02-11:
@EeliK 
would you please test the new update?
It should fix the pad/drill offset.

closed 2020-02-11 by easyw