easyw/kicadStepUpMod #39: 3D object to 2D projection is flipped

jeremyherbert, 2020-02-02:
As a learning example, I'm trying to create a footprint for the following part: https://www.sparkfun.com/products/9032 (there is a STEP model available on that page).

I started by importing the model, changed the view to the bottom of the part (using the cube in the corner) and then selected the joystick object and clicked "3D object to 2D projection". However, the resulting projection is both rotated and flipped. Am I doing something wrong here?

[image]

easyw, 2020-02-02:
@jeremyherbert 
ksu is using a FC function to make the projection... 
https://www.freecadweb.org/wiki/Draft_Shape2DView
Moreover it may be related to OCC Kernel..
so the issue should be reported at FC repo.
ATM you can workaround it simply rotating and aligning the projection.

jeremyherbert, 2020-04-08:
This is not an issue here, so closing

closed 2020-04-08 by jeremyherbert

jeremyherbert, 2020-05-08:
Just reporting on what the actual issue is and a workaround:

workaround: https://forum.freecadweb.org/viewtopic.php?f=22&t=24807&p=194839#p194816

issue: https://tracker.freecadweb.org/view.php?id=3348