easyw/kicadStepUpMod #40: Add new feature to import PCB inner layer

rniueng, 2020-02-19:
Now the Top and Bottom layers of PCB can be imported to Freecad. This is good enough for mechanical design. For some purposes, all inner layers of PCB need to be imported to 3D tool. Is there any way to do this weith StepUpMod? or could consider to add this feature.

easyw, 2020-02-19:
> For some purposes, all inner layers of PCB need to be imported to 3D tool.
@rniueng 
which is your purpose?

laurencedv, 2020-02-19:
Maybe for EM or thermal analysis in another FreeCAD workbench (I'm not the OP so I dont know, but that could be a very useful feature)

easyw, 2020-02-19:
You may need to look at @realthunder fcad_pcb tools 
https://github.com/realthunder/fcad_pcb

closed 2020-02-19 by easyw

rniueng, 2020-02-20:
One purpose is to use the exported layers to describe trace routing strategy for training and documentation. In FreeCAD, I can enable disable layer easily and make PCB (FR4) material Transparent so inner layer can be viewed. Also I can change the color for any trace.

The other thing is EM simulations. I want to analyze trace to trace inductance and capacitance.

I’m getting help from real thunder too on my installation issue. I haven’t successfully run the fc_pcb tool. That is nice tool for me too.

Thanks

获取 Outlook for iOS<https://aka.ms/o0ukef>

Closed #40<https://github.com/easyw/kicadStepUpMod/issues/40>.

Quas7, 2022-08-29:
+1
usecase: designing a PCB coil and running ElmerFEM magnetodynamic simulations on it.
Thanks for the tip with fcad_pcb. That seems to be the best route for now.