easyw/kicadStepUpMod #41: custom KiCAD PATH variable

laurencedv, 2020-02-19:
Hey,

I'm not sure if this a real issue or something I didn't understand in the StepUp Preference panes but anyhow, I'll expose the situation here and see if somebody can enlightened me:

In KiCAD on my workstations I setup a custom PATH variable pointing to the top of my lib ie:
KICADLIB = /path/to/lib-folder
Which is then nicely subsection into folder for footprint, template, 3dmodel, etc... which all point to the {$KICADLIB}/footprint ,  {$KICADLIB}/model , etc.
quite simple and easily portable, but it assume knowledge of the value behind {$KICADLIB} to complete the path to file.
Which is what I think is causing my problem when importing my pcb with the "import PCB" tool in Step-up and all the "not found model: {$KICADLIB}/bla/bla/bla"

Did I miss something somewhere? I always asked myself what was the purpose of those "ALT3DMOD" and "alternate 3d model search folder" option were or theit usable.

I can't sadly share the culprit file as it is work related but I could surely whip up something when at home (with personal project) if needed.

In any case thanks for the (what looks like (see above for current probleme :P)) a very nice tool!

easyw, 2020-02-19:
{$KICADLIB} or ${KICADLIB} ?
moreover you need to add to your StepUp preference page the path to your internal variable
i.e. if your footprint has this syntax:
    (model ${KICADLIB}/model/mymodel.wrl
then you would need to add to your StepUp preferences i.e. at the secondary 3D folder location
the absolute path to your ${KICADLIB} folder
Please have a look at
https://github.com/easyw/kicadStepUpMod/blob/master/demo/kicadStepUp-cheat-sheet.pdf

laurencedv, 2020-02-19:
Ah yes sorry I wrote from memory, ${VAR} it is.
ie: `${KICADLIB}/model/BGA/BGA6N50P2X3_150X150X68.step`

Following your step (pointing to the var value instead of the 3d model sub-folder did the trick.
Old ALT3DMOD: `/home/laurencedv/project/lib/KiCADlib/model`
new and working ALT3DMOD: `/home/laurencedv/project/lib/KiCADlib`

And finally yes I was following the steps (haha) in the quite extensive doc you have made, Im now passed import and mech constraint checker (ho boy that would have save me time 2 weeks ago -_- ...)

Thanks for the quick answer!

closed 2020-02-19 by laurencedv