easyw/kicadStepUpMod #42: Importing PCB is not loading 3D models

iharshadec, 2020-02-29:
I am trying to import KiCad PCB file in FreeCAD to generate a complete PCB 3D model.
All my 3D models are placed inside the project library itself.

> ${KIPRJMOD}/lib/pcbLibrary/kicad/shape3d/XXXX.wrl 

and 

> ${KIPRJMOD}/lib/pcbLibrary/kicad/shape3d/design/XXXX.step

When I import KiCadPCB I get this error
[image]

How to configure path so it gets all 3D model files?

easyw, 2020-02-29:
@iharshadec 
from your configuration:
> ${KIPRJMOD}/lib/pcbLibrary/kicad/shape3d/XXXX.wrl

it seems your lib is located inside your project folder....
considercthat ${KIPRJMOD} is a variable pointing to kicad project folder... 
In StepUp you can configure two 3d searching paths ...
typically are:
KISYS3DMOD and a secondary path assigned to the basic path of your external lib i.e. MY3DLIB

iharshadec, 2020-03-01:
@easyw 
Yes, I added path same way under ALT3DMOD

> D:/project/SDBreakOut_v0.1/lib/pcbLibrary/kicad/shape3d

[image]

But still having same error.

easyw, 2020-03-01:
@iharshadec 
it should work...
may you share a small pcbnew example file with this issue... even a minimal board with just one or two footprint... I would like to have a look inside the text file

iharshadec, 2020-03-02:
@easyw 
Let me know if this file is going to work out for you.
[SD3DTest.zip](https://github.com/easyw/kicadStepUpMod/files/4275411/SD3DTest.zip)

easyw, 2020-03-02:
@iharshadec 
the issue you are having is related to the path of your 3d models...
you need to have both 'model.step' and 'model.wrl' in the same location..
StepUp (and also the kicad internal STEP exporter) is switching from wrl to step to assembly the board.
Just move your design files to the same place of wrl files, and your board will be loaded in FreeCAD directly.
In your example you even don't need to declare an extra variable path, because the models are located relatively to project folder (KIPRJMOD variable)

iharshadec, 2020-03-03:
@easyw 
Yes, That worked out. Thank you 
I have to take care to have same file name and same folder for both .step and .wrl files.
For silkscreen I see that it needs DXF files but tracks should be there directly imported - correct?

No tracks here.
[image]

This is file.
[image]

easyw, 2020-03-03:
for tracks you need to use a different button ... please have a look at the the [kicadStepUp-cheat-sheet.pdf](https://github.com/easyw/kicadStepUpMod/blob/master/demo/kicadStepUp-cheat-sheet.pdf) page 12.

iharshadec, 2020-03-03:
Cool ! 
Thank you @easyw

closed 2020-03-03 by iharshadec