easyw/kicadStepUpMod #43: Issue with Importing 3D Parts

bricechart, 2020-03-12:
I have been using kicadStepUpMod for several projects.  I recently upgraded my PC and redownloaded FreeCAD and kicadStepUpMod and ran into a new issue.

[image]

If I roll back kicadStepUpMod to the 2/5/19 commit, the problem appears to goes away.  The problem is that many of the STEP files I use are from part manufacturers.  My PCB design is showing many files that have this "NOT fused" error.

easyw, 2020-03-13:
@bricechart 
in recent releases of StepUp there is an option to choose the 3D model loading mode...
your settings probably are with NotMultiParts option... 
Please always specify also your OS and FC versions. FC0.18.4 or FC0.19-daily are best options
[image]

bricechart, 2020-03-15:
Thanks for the tip.  For me, that still doesn't solve the problem.  When using Hierarchy, and Standard I still get the error.  When I try "Simplified" it appears to work and the whole program crashes.  This was all the FC 18.4.

I tried the latest nightly (0.19 - 20036) and ran into a slightly different issue.
[image]

easyw, 2020-03-15:
hi @bricechart 
this is different issue.. I would suggest you not to use 'iges' models... iges is an old standard mostly abandoned... The best option is to have the library in 'STEP' format.
In FreeCAD 0.18.4 Hierachy option is not working, and then the importer will switch automatically to Compound mode.
In FC0.19 Hierarchy should work correctly even with your iges model...
Would you mind to upload a simplified board with i.e. two footprints and the 3d models which are giving you this issue? Without the 3d models and a testing board it is really difficult to help in solving the issue.

bricechart, 2020-03-15:
Thank you for the quick response and explanation. I've attached the board and 3D CAD files as suggested.  Hopefully you can use this to recreate the issue.

[3D Issue.zip](https://github.com/easyw/kicadStepUpMod/files/4334119/3D.Issue.zip)

bricechart, 2020-03-15:
Thank you for the quick response and explanation. I've attached the board and 3D CAD files as suggested. Hopefully you can use this to recreate the issue.

[3D Issue.zip](https://github.com/easyw/kicadStepUpMod/files/4334127/3D.Issue.zip)

easyw, 2020-03-15:
@bricechart 
please have a try with the attached project.. I simply have loaded the 3d iges model in FreeCAD 0.19 and saved it as STEP file, then updated the pcb accordingly and the pcb can be loaded in FreeCAD 0.19 correctly.
I will have a better look at the iges problem, but for a better workflow I would suggest you to avoid iges and stay on step format.
[3d-issue-solved.zip](https://github.com/easyw/kicadStepUpMod/files/4334462/3d-issue-solved.zip)

At the same time I confirm the issue for multipart on FC 0.18... I'm going to investigate further

bricechart, 2020-03-15:
Great, thanks for the update.  I don't think I have too many parts that are IGES.  I will just convert them and keep working with what I got.  Thanks for all your help.

easyw, 2020-03-19:
this https://github.com/easyw/kicadStepUpMod/issues/43 should solve 'iges' importing... minor issue model will be loaded without colors

closed 2020-03-19 by easyw