easyw/kicadStepUpMod #46: footprint missing one pad

jeremyherbert, 2020-04-08:
I'm using this addin with FreeCAD 0.18-16146. When I try to export a footprint from the attached file, I am missing one pad and I am not sure why?

[image]

[footprint.FCStd.zip](https://github.com/easyw/kicadStepUpMod/files/4448260/footprint.FCStd.zip)

easyw, 2020-04-09:
@jeremyherbert 
the converting algo is quite simple and requires a strait sequence of edges.
The missing pad has a wrong edge sequence.
Anyway I would suggest you to use the rectangle tool to create pads instead of placing 4 lines together
[image]
.
Here the fixed file.
[footprint-fixed.zip](https://github.com/easyw/kicadStepUpMod/files/4458705/footprint-fixed.zip)

jeremyherbert, 2020-04-10:
Thanks! I actually created this using a DXF import and then the DXF to sketch feature. Is there a way to have this convert the lines into rectangles?

[pads.dxf.zip](https://github.com/easyw/kicadStepUpMod/files/4459420/pads.dxf.zip)

easyw, 2020-04-10:
@jeremyherbert 
have you used the StepUp button to convert the imported dxf shapes?
[image]

for me, using this button, it just worked
[pads-dxf-to-sketch.zip](https://github.com/easyw/kicadStepUpMod/files/4460063/pads-dxf-to-sketch.zip)

jeremyherbert, 2020-04-10:
Yes, that is the button I used. I tried it several times too... I will try to investigate further

jeremyherbert, 2020-04-11:
I just tried this again:

1. Create a new document in FreeCAD
2. File => Import, browse to the `pads.dxf` file and open
3. Select all of the entities that were imported and click the "2D object (or DXF) to Sketch" button
4. Delete the entities, leaving just the sketch behind
5. Rename the sketch to "Pads_TH_SMD"
6. Select the sketch and click "Footprint editor and exporter"
7. Save the mod file
8. Open in Kicad 5.1.5

This time it has given me different missing pads? I think I am going to try to install the freecad beta as it seems you are running a newer version that I am 

[test.kicad_mod.zip](https://github.com/easyw/kicadStepUpMod/files/4463881/test.kicad_mod.zip)
[new.FCStd.zip](https://github.com/easyw/kicadStepUpMod/files/4463882/new.FCStd.zip)

[image]

jeremyherbert, 2020-04-11:
Ok, it looks like when I use the freecad version from here: https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre it has worked fine three times in a row. So I suppose this is due to a bug in freeCAD.

Shall I close this issue, or is it better to wait until 0.19 is released?

easyw, 2020-04-11:
We can leave it open until 0.19 will be released as stable. 
Thx

jeremyherbert, 2020-04-24:
I have unfortunately encountered this problem again, even using the latest freecad. Could you point me toward the section of the code that does the pad detection? I could perhaps try to fix this myself.

This file has the center pad missing when converted to a kicad footprint.

[pads.dxf.zip](https://github.com/easyw/kicadStepUpMod/files/4526029/pads.dxf.zip)

easyw, 2020-04-25:
@jeremyherbert 
the routine is called

closed 2020-05-07 by easyw