easyw/kicadStepUpMod #47: Export footprint

jneiva08, 2020-04-30:
I'm trying to make footprint of [WE-TPC-4828](https://www.we-online.de/katalog/datasheet/744043101.pdf) and I'm guiding from your video https://youtu.be/5IcAyfhaf2g .
Have made previously step file with the recommended land pattern.

When try to export gives following error:

Running the Python command 'ksuToolsFootprintGen' failed:
Traceback (most recent call last):
  File "/home/xathar/.FreeCAD/Mod/kicadStepUpMod/kicadStepUpCMD.py", line 1920, in Activated
    kicadStepUptools.PushFootprint()
  File "/home/xathar/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 16430, in PushFootprint
    export_footprint(name)
  File "/home/xathar/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 17029, in export_footprint
    drl_found=collect_drl(pth_ordered)
  File "/home/xathar/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 17673, in collect_drl
    px=(p[0][10].x+p[2][11].x)/2;py=(p[0][11].y+p[2][11].y)/2;

What I'm doing wrong?

easyw, 2020-04-30:
@jneiva08 it is really difficult to say anything without having a FC file

jneiva08, 2020-04-30:
thanks for sending the file.
suppose it isn't change only the name.
Have any guide to how make it?

i don't have any experience with 3d software

easyw, 2020-04-30:
@jneiva08 
have a look at here 
https://github.com/KiCad/kicad-footprints/pull/972

closed 2020-05-03 by easyw