easyw/kicadStepUpMod #49: SilkS layer out of the board

leoheck, 2020-05-24:
@easyw easyw, this plugin is getting better and better. This is awesome, thank you for that.

This is one thing that I would like to give as the feedback since some time ago.

What do you think to have something that cuts the silkscreen (removing parts that are not over the board). Please, have a look at the following example.

[image]

I don't know if this is too much of work since right now, I was not able to import the SilkS in the right position in the first place. But you can tell this better.

easyw, 2020-05-24:
> I was not able to import the SilkS in the right position in the first place.

Please have a look at here: https://forum.kicad.info/t/stepup-silkscreen-in-wrong-location/22471/3

> What do you think to have something that cuts the silkscreen (removing parts that are not over the board).

the wanted result could be obtained in FC with a common boolean operation between Silks and pcb, but this is a bit out of scope here
The best for this option is to be implemented directly in KiCAD ...

closed 2020-05-24 by easyw

leoheck, 2020-05-24:
Cool thanks. I was also doing this on Freecad, it is easy, but it is better if it can be automated, right? I will file a feature request on Kicad.

leoheck, 2020-05-25:
This issues was already there on kicad repo.
https://gitlab.com/kicad/code/kicad/-/issues/2293