easyw/kicadStepUpMod #50: Export .step as part instead of being an assembly.

leoheck, 2020-05-25:
I am trying to export a .step file of my board as a part. When I use KicadStepUp exporting the board from the Freecad it exports like an assembly being a modular object. 

Does the KicadStepUp have any way to export the board as a part?

I want to prevent the PCBs being exploded when used in other assemblies. This is a view of .step from the GrabCad. I want to avoid that if possible. Do you know if there is a way to do that?

[image]

leoheck, 2020-05-26:
I saw that the plugin has a button to check if the step will create a multi solids object.
I think it is related. Ths one is telling me this.

Do you know what causes this and how can I make it a single solid?

[image]

easyw, 2020-05-26:
> I am trying to export a .step file of my board as a part. When I use KicadStepUp exporting the board from the Freecad it exports like an assembly being a modular object.

The process to create a single object from an assembly is a boolean process, which can be very complex.
Particularly if you want to add tracks and silks to your board, it will not be possible to make a union that will remain as a single object because the pcb and the tracks/silks are detached... then even if you make a union (try to make it) the result will be a compound and not a single object.
In general to have a single union of objects, they need to be attached each others, otherwise they will result in an assembly of objects (called compound).
The best option is to keep the model as a hierarchy (board, pcb, components, tracks, silks) inside a multilevel containers.
The CAD which will load the model should parse it correctly... if not, then the issue is on the reader/parser in the loading file CAD.
You can test loading the model i.e. in CADAssistant of OCC

> I saw that the plugin has a button to check if the step will create a multi solids object.

that button is a tool to check if the component can be modelled as a single unioned part and it is most useful for building the kicad 3d library.

leoheck, 2020-05-26:
So, as far as I understood

To achieve that, a boolean operation of the parts should be performed. If it can't be done then the output won't be this single solid I want, right?

1. What makes the operation to fail?
2. It is a broken or a not good part in the pcb?
3. If I remove/replace that part, is it possible to make it a single part?
4. Does the KicadSketchUp have any tool to report a list of parts that can be problematic?

leoheck, 2020-05-26:
I have 3 boards that are single solid (considering .step generated by Kicad and also Generated by KicadStepUp)

I have 2 boards that cannot be merged. I have no idea why! I already tried thousands things. I have no idea where this issue arises.

- I removed almost everything (kept only one resistor is on top)
- Changed/Recreated the board Outline
- The Kicad_pcb was recreated

easyw, 2020-05-26:
> I have 2 boards that cannot be merged. I have no idea why! I already tried thousands things. I have no idea where this issue arises.
> 
>     * I removed almost everything (kept only one resistor is on top)
> 
>     * Changed/Recreated the board Outline
> 
>     * The Kicad_pcb was recreated

If you have TH components, i.e. a simple resistor, normally you cannot make a real union of pcb and resistor because the resistor is detached from the board (the leads don't touch the pcb, because the holes are greater than the leads)

leoheck, 2020-05-26:
> If you have TH components...

Ah, good info here. So, if I overlap the 3d model with the PCB by changing the THT placement (it won't be beautiful and precise) but I might achieve something I want (considering I am not using 3d models with issues)

> that is why I forced KiCAD 3D library to be built with a high level of checking/quality...

What does this mean? Does it mean that the .step exported from KicadStepUp is better than one exported from the Kicad?

There are any settings to skip 3d models that are not good?

easyw, 2020-05-26:
> What does this mean?

simply that the kicad packages3d have been built keeping a high level in the quality of the models

leoheck, 2020-05-26:
Ah, sure. Kicad stuff it the best. For sure, it is the best tool I use.

I  have some models that I got from the internet.
That I removed to validate this idea.

However, using only Kicad models I was not able to reproduce this (boolean-safe thing) in one board. I will remove the THT components keeping only Kicad models as a test. To see if it works for that particular board.

leoheck, 2020-06-02:
I am not understanding this issue. I made a lot of tests. I still have no clue what is going on. 

For instance, I have 2 boards:
1. `Board_A` small with LEDs and resistors (SMD) and 1 THT header, when importing, KicadStepUp tells me that this has Multi Parts. Then removing all components but a Resistor or other SMD, and **it still does not behave a solid part** when exported. Also, I tried the `3D Loading Mode` as `NotMultiparts`. Nothing has improved.

2. `Board_B` bigger, with many SMDs, 1 THT header, 1 THT connector, and 1 THT Buzzer.
KicadStepUp tells me that are some icons missing but the exported .step model **is solid part**.

Also, I tested the `Check if the selected part would be exported to STEP as a single solid` and it says the same for both boards. 

[image]

I have no idea what makes the board not be exported as a solid part.
Can I fix it by removing parts in Freecad? Do I need to configure something special on Kicad/Freecad?

leoheck, 2020-06-02:
Ha! This looks interesting.. I was not expecting something like this. Speccially when you said this before (here https://github.com/easyw/kicadStepUpMod/issues/50#issuecomment-634233280)

> simply that the kicad packages3d have been built keeping a high level in the quality of the models

[image]

[image]

[image]

[image]

[image]

leoheck, 2020-06-02:
This is another test.
Here, I was able to export it as a single solid but I don't know why?
No operations were made inside the Freecad.

`Top-Right` - Exported from Freecad with tracks. As you said, this won't be a single solid. 
`Top-Left` - STEP Exported directly from Kicad
`Bottom-Right` - Exported from Freecad after importing the Kicad STEP
`Bottom-Left` - Exported from Freecad after importing with KicadStepUp 

You also said before:
> If you have TH components, i.e. a simple resistor, normally you cannot make a real union of pcb and resistor because the resistor is detached from the board (the leads don't touch the pcb, because the holes are greater than the leads)

It does not behave as you expected.

[image]

leoheck, 2020-06-02:
My conclusion so far. **I have not idea what is going on.** 

If you have any idea to understand this. Please enlighten me. :)

easyw, 2020-06-02:
> I am not understanding this issue. I made a lot of tests. I still have no clue what is going on.
> 
> For instance, I have 2 boards:

@leoheck simply post your BoardA and BoardB pcbnew files and I would help to solve your riddle...

leoheck, 2020-06-02:
Ok, fair enough. Also, thank you for your help in advance.

The board that is not working, is this one attached here [board_a.zip](https://github.com/easyw/kicadStepUpMod/files/4720124/board_a.zip)

Some considerations about this board, if needed
- This `WS2812B_5050.step` the model was scraped from the internet 
- Other models are from the Kicad library.
- There is one Model Missing (it is ok, it was from the internet too, and now I removed it)

Models can be replaced if they are not good. My biggest issue is that I have no idea how to have a working board yet. I have no idea what should I do to guarantee it will always be a single solid. 

So, I don't want the solution for this board itself. 

**I am interested in the path to the solution to replicate it to all of my boards and also to teach my colleagues how to generate something like a single solid board.** 

If you can find the issue and explain the process for me I would be very glad. :)))

```
Application: Pcbnew
Version: 5.1.6-c6e7f7d~86~ubuntu20.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3
Platform: Linux 5.4.0-34-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.68.0
    Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON
```

Also, extra info, just to make sure you have enough tools to investigate this.

KicadSetupUp version installed is 8.1.0.8

Current Freecad STEP Export/Import Configs
[image]

[image]

easyw, 2020-06-03:
@leoheck 
and your FreeCAD full version?

easyw, 2020-06-03:
a small tuto 
.
[image]

leoheck, 2020-06-03:
Freecad version. Forgot that, my bad.
Thanks for this tutorial. I will test this and come back with the result.

```
OS: Ubuntu 20.04 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21280 (Git) AppImage
Build type: Release
Branch: master
Hash: 6f3160db3e88733536c7eaf97ad7d6ebd21baccd
Python version: 3.8.2
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
```

easyw, 2020-06-03:
@leoheck 
which is the viewer you are using for your 3dmodels (as per your screenshots)?
KSU version should be 9.6.0.2

leoheck, 2020-06-03:
I am using GrabCad, do you know it?
It is free to create an account there and it is really good for collaboration.
https://workbench.grabcad.com/

I was testing this, there right now.
And your example is awesome. I was able to produce something a solid board.

I am trying to explode the board there, and it is a single solid. And it is colored. 
It is perfect!

Thank you so much for your time and patience.

leoheck, 2020-06-03:
I got the info from here, maybe it is not updated.

[image]

leoheck, 2020-06-03:
Ah, here it is. 
So, maybe it is good to fix the `KSU Help Tips` page too

[image]

easyw, 2020-06-03:
The version in help is wrong.. I updated it... when you launch the ksu tool (not the wb) you will see it in the report panel

easyw, 2020-06-03:
I don't use Grabcad or similar because cloud service won't leave you the control and property of your own work and data.
Moreover it seems Grabcad has a very bad CAD step interface because it is not able to handle hierarchical Step models.

leoheck, 2020-06-03:
> BTW the ksuWB version and the ksuTools version are two different versions.. 

Got it!

> cloud services won't let you have the control and property of your own work and data.

This is good advice!

I use it because I have to, for one project. So it is not my choice, unfortunately.

closed 2020-06-04 by leoheck

leoheck, 2020-06-04:
Cool, this method for that small board works.
Now I have another board with many more parts.

Importing with KSU it gives me this:
``` 
3D STEP model TL331XX is WRONG
or are not allowed Multi Part objects...
@ /home/lheck/myboard/pcb/library/3dpackages/TL331XX.step
...stopping execution! 
Please fix the model or change your settings.
``` 

Then, removing this from the board.
Importing it with KSU again (no warning/error messages are displayed)
But after `Make a union of selected objects`
Gives me this:

[image]

So, is it possible that KSU can't always detect when the step has are multi parts?

leoheck, 2020-06-04:
Maybe floating objects can cause this. But how to debug since all of them looks touching the surface.

easyw, 2020-06-04:
@leoheck 
1) in your ksu settings you have chosen: '3D loading: -> Not Multipart', so this is why you get the error in loading TL331XX, which is probably a muti-part object
2) a boolean union of parts is a real complex operation for a 3D CAD... not always it will succeed... have a look at FreeCAD forum for this topic and you will found a very large subject :smile:
(https://wiki.freecadweb.org/Part_Boolean -> Coplanar problems)

leoheck, 2020-06-04:
Yes it is. 
I was able to find the culprits. They were 5 models. KSU couldn't tell me these models where multipart. But KSU helped me to find them.
It was many iterations, removing components, importing boards and performing Boolean operations. What a work. 😅

easyw, 2020-06-05:
@leoheck when loading a pcb in FC, ksu is listing all the models in report view;
if you simply copy the list from the report panel into a text editor and search for 'multi Part found!' there, you could simplify your searching procedure and reducing the time for polishing your board.

leoheck, 2020-06-06:
Ah, good idea. This didn't pass through my mind since KSU didn't show the multi-Part error popup.

leoheck, 2020-06-07:
> if you simply copy the list from the report panel into a text editor and search for 'multi Part found!' there, you could simplify your searching procedure and reducing the time for polishing your board.

Ok, but this does not work all the time. Now I have a board that doesn't have any multi-part (at least no popup was shown with that, and any info with that was presented in the console) the board was made a solid. 

This part here fails but it is silent, It is hard to understand what is happening.
I was testing the .step AP214
https://www.3dcontentcentral.com/download-model.aspx?catalogid=171&id=441404

This part is from Texas Instruments and it also fails but it is not silent. 
Downloaded from here.
https://vendor.ultralibrarian.com/ti/embedded?vdrPN=TPS79633QDCQRQ1&sid=0172051bc31f000cbe2b4466084703069017c06100fb8&c=1
```
3D STEP model SOT-223-6 is WRONG
Running the Python command 'ksuToolsOpenBoard' failed:
Traceback (most recent call last):
  File "/home/lheck/.FreeCAD/Mod/kicadStepUpMod/kicadStepUpCMD.py", line 656, in Activated
    kicadStepUptools.onLoadBoard()
  File "/home/lheck/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 8012, in onLoadBoard
    Load_models(pcbThickness,modules)
  File "/home/lheck/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 5591, in Load_models
    stop   

name 'stop' is not defined
```

leoheck, 2020-06-07:
For the first one SOT-223-6 I made one now, to avoid this issue. 
Kicad library could have it since it has only these:
SOT-223-6
SOT-223-6

easyw, 2020-06-08:
> For the first one SOT-223-6 I made one now, to avoid this issue.
Kicad library could have it since it has only these:
SOT-223-3
SOT-223-4
SOT-223-5
Do you know where I can make a request?

[FCAD_script_generator/GW_QFP_SOIC_SSOP_TSSOP_SOT/cq_parameters_sot.py#L320](https://github.com/easyw/kicad-3d-models-in-freecad/blob/e935e68d06e78835456fe100f5a70a9a926a2a45/cadquery/FCAD_script_generator/GW_QFP_SOIC_SSOP_TSSOP_SOT/cq_parameters_sot.py#L320)