easyw/kicadStepUpMod #52: Can't export poly pad with oval hole

sambazley, 2020-07-11:
I'm creating a footprint for a USB port that requires two poly pads, each with an oval hole. The footprint fails to export with:

```
Running the Python command 'ksuToolsFootprintGen' failed:
Traceback (most recent call last):
  File "/home/user/.FreeCAD/Mod/kicadStepUpMod/kicadStepUpCMD.py", line 1920, in Activated
    kicadStepUptools.PushFootprint()
  File "/home/user/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 16551, in PushFootprint
    sk_d=Draft.makeSketch([wns,wnc[0]])

list index out of range
```

[usb.zip](https://github.com/easyw/kicadStepUpMod/files/4906203/usb.zip)

easyw, 2020-07-12:
@sambazley 
you need to make the pads with a workaround
here the files
[USB_Micro-B-modified.zip](https://github.com/easyw/kicadStepUpMod/files/4909351/USB_Micro-B-modified.zip)

sambazley, 2020-07-12:
That works, but it results in one of the poly pads covering the hole in the 3D viewer.
[image]

It displays correctly on the back of the board though.
[image]

I tried to fix this by using a poly pad that doesn't cover the hole, but this pad doesn't show up at all in Kicad.
[image]

[usb.zip](https://github.com/easyw/kicadStepUpMod/files/4909661/usb.zip)

Thanks for your help.

easyw, 2020-07-13:
I just forgot to remove solder paste on one SMD pad:
[usb-micro-b-v2.zip](https://github.com/easyw/kicadStepUpMod/files/4910755/usb-micro-b-v2.zip)

sambazley, 2020-07-13:
Oh I didn't think to check that. Thank you :)

closed 2020-07-13 by sambazley