easyw/kicadStepUpMod #53: Error pushing edges

tclarke, 2020-07-26:
can't push edges 
kicadStepUptools.py:15645 currentTextChanged not a member of QComboBox.
Probably needs to be editTextChanged

tclarke, 2020-07-30:
> @tclarke
> would you mind to upload the FC file which gives you the error?
> could you please add also FC full information?

Downloaded a fresh copy when initially testing so 0.18.4. Don't recall if that was the version I originally saw the problem but it was no more than a release behind. Windows 10 64-bit.

easyw, 2020-07-31:
@tclarke so your issue can be closed?

tclarke, 2020-07-31:
The pull request I sent fixes this issue.

easyw, 2020-08-01:
@tclarke 
would you please attacha a FC file wich is causing this issue? (just zip it and attach it to a essage)

tclarke, 2020-08-05:
sorry, just noticed this...I'll put something together

easyw, 2020-08-24:
@tclarke any news on it?

closed 2020-09-09 by easyw