easyw/kicadStepUpMod #55: local variable 'firstCoordinate' referenced before assignment

morganrallen, 2020-07-27:
Trying to create a complex footprint but fails with `local variable 'firstCoordinate' referenced before assignment`

Even simplified forms fail with the same error.

[humming_pad.FCStd.gz](https://github.com/easyw/kicadStepUpMod/files/4983624/humming_pad.FCStd.gz)

morganrallen, 2020-07-27:
I've tried various `Pads_TH_SMD`, `Pads_Round_Rect` but all have the same error. FWIW it would be super handy to have a complete list of the supported Sketch name tags and what they're for.

easyw, 2020-07-30:
@morganrallen 
for this 'artwork' pad type I would suggest to discretize the sketch (with ksu tool) and then export it as a polyline pad
[humming_pad-discretized.zip](https://github.com/easyw/kicadStepUpMod/files/5002747/humming_pad-discretized.zip)

> FWIW it would be super handy to have a complete list of the supported Sketch name tags and what they're for.

I would accept a user manual contribution, based on the example included with the WB 😉

closed 2020-09-09 by easyw