easyw/kicadStepUpMod #56: Feature request : move footprints

pwnorbitals, 2020-07-28:
I would like to move my footprints on my board, using the FreeCAD Sketcher, in the same way I can place the edge cuts.

Is this feature already implemented / planned / dropped ?
If it is planned, what would be the best way to help implement it ?

Thanks

easyw, 2020-07-30:
@pwnorbitals 
this is already implemented...
you need to use the push-pull tools for pulling in FC the model (footprint) positions and then moving to the right location and then pushing the new locations back to KC
a demo on utube
https://youtu.be/6R6UEUScjgA

pwnorbitals, 2020-07-31:
I am aware of this. What I need is being able to place exactly the position of the footprints using the 2D sketcher and the constraints system

easyw, 2020-07-31:
Your footprint and then your 3d model can be aligned precisely using i.e. Aligner of Manipulator WB... then you can align precisely some reference of your 3d model (related to its footprint) to any external reference. This is the only way to proceed ... align the 3d model to your i.e. mechanical constrains, and then update the footprint accordingly.

pwnorbitals, 2020-08-03:
What if I don't have a 3D model for my footprint ? I don't have use for it, and it would take a lot of work to make

easyw, 2020-08-03:
@pwnorbitals 
> What if I don't have a 3D model for my footprint ?

you should add a dummy 3d model i.e. a cylinder of 1mm radius

> Also, your technique doesn't allow for complex circular symmetries or any of the advanced Sketcher functionnalities afaik

Why? if you use the the cylinder as a reference for your sketch, you can apply any advanced Sketch functions to the cylinder center of mass and create your design accordingly.

pwnorbitals, 2020-08-04:
Thank you for taking time to answer

closed 2020-08-04 by pwnorbitals