easyw/kicadStepUpMod #58: Automate WRML & STEP fabrication

Franck78, 2020-09-08:
From reading different PR on 3dmodels, I noticed following things:

* lot of PR  are aborted cancelled because the 3d contains a not editable included part
* not fusionned .STEP request
* or "filename" doesn't match object name
* ....

That being said, it is tedious to obtain the VRML and the STEP from the model displayed in the FreeCAD window.

I would like to see a single button "Create files for KiCad 3d library" handling everything.

If you look at the 3d model here https://github.com/KiCad/kicad-packages3D/pull/632 it is
* not merged for a strange reason
* is the perfect Cad everyone dreams of

This is drawing is fully parametric. If you want to generate the 5mm button, you enter 5 (not easy to find where).

The "Create files for KiCad 3d library" could integrate a basic loop where it iterates for a list of values. For this SW_SPST, it would be " 3.8 5.0 8.0" applied to a selected constraint.

In a single click, generates all variants , all files, superchecked, error free...

note: variant size for this SW, see the loop here ;)

6*6 SMD Micro Switch Height Table
--
4.1 | 4.3 | 4.5 | 4.7 | 5 | 5.5 | 6 | 6.5 | 7 | 7.5 | 8
8.5 | 9 | 9.3 | 9.4 | 9.5 | 10 | 10.5 | 11 | 11.5 | 12 | 12.5
13 | 13.5 | 14 | 14.5 | 15 | 15.5 | 16 | 16.5 | 17 | 17.5 | 18
18.5 | 19 | 19.5 | 20 | 21 | 22 | 23 | 24 | 25

easyw, 2020-09-09:
@Franck78 
you have two options IMO:
1) use scripts to create your 3D models
2) have a closed look at kicad StepUp to help in exporting your FreeCAD design

Franck78, 2020-09-09:
Are you saying that there is
* export as .step
* export for Kicad  (wrml+ fusioned step)

?

easyw, 2020-09-10:
@Franck78 
yes ... have a look at kicadStepUp-cheat-sheet.pdf
(it is available here at [the repo](https://github.com/easyw/kicadStepUpMod/blob/master/demo/kicadStepUp-cheat-sheet.pdf) or directly from FreeCAD in the Demo Menu of the StepUp WB)
There is a button to export exactly what you need ... moreover there is also a button to check if the exported file will be a single unioned file when re-imported

Franck78, 2020-09-10:
First time a clicked on the the pdf, Freecad displayed it in raw format. 

Next, tooltip choose for the function we are speaking of is wrong.
 * Export 3dmodel to KiCad
Well no, it is more "Create WRML & STEP". Can be for Kicad or this repo or anything else, you don't know.

For some reason, the tooltip is more accurate in the "Ksu tools windows" ! 

I love to say my computer is good at repeating stupid tasks. Answering same questions again & again is a stupid task. I suggest you create a body for all the answers. Including the most annoying of all "File exists, overwrite ?" It is yes in 99.999% ;)
And save that new body inside the FCstd file.

> Select ONE single part object !
> suggestion for multi-part:
>   Part Boolean Union (recommended)

This forces to modify the model to suit your need. Simply not good. Just silently Union before writing files.

This function is the strict equivalent of what we found in main_generators.py & cq_model producing bodies.

 It just don't have the "iterate" yet. 

Please open the Molex 5267 3d design. It is denying the purpose of FreeCAD to try to describe this thing with script's primitives.
Now what if the designer do it with three bodies. Left, middle, right. With little help, this script can generates the entire serie, one click.

[image]

easyw, 2020-09-10:
@Franck78 
1) you miss that the WB started as a single FC Macro late in 2015 when FC didn't even had Body feature
2) to force the user to modify the object in a single unioned part is the right action to obtain a real single object; sometimes boolean operation will create a single object but that will create a multi part when exported to STEP (you need to go deeply in STEP format to assimilate that concept)
3) if you, after 5 years of improvements in FC features and python functions, are ready to rewrite all the generators and are willing to check if they work correctly in many releases of FC and in all the platforms (OSX, Win Linux), as well as check if the generated models are all compliant with the OS platforms and CAD sw, I will be glad to merge the code (obviously python friendly and formatted generator by generator)
3) the same concept for KiCAD StepUp and its help tooltips and user manual

>Export 3dmodel to KiCad
Well no, it is more "Create WRML & STEP". Can be for Kicad or this repo or anything else, you don't know.

No, kicad StepUp has started exactly to export a FC model to be the most compatible with kicad 3D viewer... and yes I do know it 😜

Franck78, 2020-09-10:
point2: you need the output unionned. Not the design of the user. What if the user performs the operation "Make a union of selected objects" , do the export then say yes routinely to "Save before quit" ? Screwed ! Unionned for good, bye bye bodies.

point3: I'm not speaking here of the scripted generators. I'm saying they don't ask questions, do the job from 'bodies' and this function should do the same minus iteration.

easyw, 2020-09-10:
point 5: fork the code and develop your fork

closed 2020-09-10 by easyw