easyw/kicadStepUpMod #60: multiple pads in one Pads_Poly sketch

sborshch, 2020-10-22:
Hi. I'm trying to create solder jumper consist of two semi-circle pads. I's my first attempt to create something in FreeCAD. I want both pads to be equal and size be defined by single radius constraint. I didn't find any way to make equality constraints between two sketches so I desided to place both pads into one sketch (I saw multiple rectangular pads in one sketch Pads_TH_SMD in footprint-complex-arc-pads example), so I created sketch with two pads, named it "Pads_Poly", but unfortunately ksu Footprint generator generates footprint with only one pad. Am I doing something wrong or is it impossible to define multiple pads in one Pads_Poly sketch?

Here is my jumper FreeCAD file.
[jmp.zip](https://github.com/easyw/kicadStepUpMod/files/5425659/jmp.zip)

By the way, removing small circles from pads result in generator error "list index out of range".  Now I understand that those circles acts as pad anchors, but I think script should display more descriptive message.

easyw, 2020-10-24:
@sborshch 
unfortunately ATM the only way is making a sketch each polyline pad... in FC you can refer to external geometry to keep references or constrains.

> By the way, removing small circles from pads result in generator error "list index out of range". Now I understand that those circles acts as pad anchors, but I think script should display more descriptive message.

I know that my tools are lacking of documentation. 
You may consider to contribute with a PR, adding tips to the main doc:
https://github.com/easyw/kicadStepUpMod/blob/master/demo/kicadStepUp-cheat-sheet.odg

luzpaz, 2022-04-30:
Ticket should be labeled 'documentation'