easyw/kicadStepUpMod #61: Option to re-import models in same file

s-ol, 2020-10-23:
When iterating on footprints and a PCB at the same time, I would like to be able to pull the position of models from KiCAD into FreeCAD, but it would be nice to also be able to update the models and PCB. A full reimport (delete everything that was imported, reimport) would be okay, the only problem is that the "Load KiCAD board" feature creates a new document, so any non-pulled content is also lost.

easyw, 2020-10-24:
@s-ol 
already available using Import instead of Open in File Menu

closed 2020-10-24 by easyw

s-ol, 2020-10-26:
@easyw: the import option does not update old models, and it imports the PCB in a different Hierarchy / Structure than the Open option / load Kicad board does. re-importing a board after opening it previously also doesn't work as expected:

Here I first opened the board using `File > Open`, which produced the top-level Part "hex", with all 3d models loaded and placed correctly on the board (upper right in the 3d view).
Then I imported the board again using `File > Import`. This created a bunch of empty top-level parts, a new top-level sketch and a new PCB part inside the old top-level Part "hex". The new PCB part is not placed according to the auxiliary axis origin as specified in the settings, and none of the 3d models have been updated, imported or moved.
 
[image]

Either this is not working as intended, or this feature is not able to accomplish what I asked about.

easyw, 2020-10-28:
@s-ol 
I need to understand your need..
1) if you need to import your pcb in a previous FC file in which you have i.e. an enclosure, you need to delete the old pcb and import the kicad pcb in the FC file
2) if you need to update the pcb and/or the 3d models to a previously imported pcb, you need to use instead the button to pull the sketch (which allows you to substitute the previous pcb) and/or the button to pull 3D models placement to pcb (both are described in StepUp cheatsheet).

s-ol, 2020-10-28:
@easyw I should have taken more time to try what you suggested last time, after giving it another shot I figured out how I can do what I want with the Import feature after all. After your first comment, I thought that the Import feature would work to *update* the existing PCB, which it does not. In fact it seems like using the Import feature while the same PCB is already loaded creates unexpected results; at least to me the results posted above don't follow any logic or expectation.

However by deleting the whole imported PCB and then reimporting I get what I want: a fresh board import with the latest models, PCB outline and footprint placement.

Something to note in terms of documentation is that I was not at all aware that the KicadStepUp (or any workbenches for that matter) added functionality outside of the workbench and workbench toolbars/menus themselves, so it didn't occur to me to try using Open or Import directly. Maybe it would make sense to change the behaviour of the toolbar button "Load Kicad PCB" to act like the Import feature does unless there is no active document? After all the Import feature is the most general and one can emulate the current behaviour by creating a new document first.