easyw/kicadStepUpMod #62: Outline failing to export to PCB

Shackmeister, 2020-11-02:
Attached outline gives the following error when trying to export to PCB

[KSU_Outline.zip](https://github.com/easyw/kicadStepUpMod/files/5475285/KSU_Outline.zip)

`14:19:12  Running the Python command 'ksuToolsPushPCB' failed:
Traceback (most recent call last):
  File "/home/shack/.FreeCAD/Mod/kicadStepUpMod/kicadStepUpCMD.py", line 893, in Activated
    kicadStepUptools.PushPCB()
  File "/home/shack/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 15719, in PushPCB
    export_pcb(name,SketchLayer,skname)
  File "/home/shack/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 19795, in export_pcb
    if hasattr(mypcb.setup, 'edge_width'): #maui edge width
  File "/home/shack/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 936, in __getattr__
    raise AttributeError('{}: key "{}" not found'.format(self._line,name))

1: key "setup" not found`

14:17:28  PoM not present
14:17:28  using 'Part' container
14:17:28  FC Version 019-0
14:17:28  kicad StepUp version 9.7.1.0

easyw, 2020-11-03:
no issue on:
OS: Linux Mint 20 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Branch: unknown
Hash: 1f0b7793e6e48c18e818f17b082790a4af0124d6
Python version: 3.8.5
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.3.0

and kicad StepUp version 9.7.1.0

(the same with AppImage
OS: Linux Mint 20 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22894 (Git) AppImage
Build type: Release
Branch: master
Hash: 9eb080488d970d313c538473e7272117ea0a7cd1
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
)

easyw, 2020-11-04:
@Shackmeister 
IIRC this issue is related to a not clean FC installation, with a mixed python environment (typically coming from a system upgrade).
I'm closing it. Feel free to re-open it in case of need.

closed 2020-11-04 by easyw