easyw/kicadStepUpMod #63: Command line tools to generate rendered images of the PCB

leoheck, 2020-11-18:
Hey @easyw do you have any tool to generate a rendered image from Kicad's boards on a command line or on a script?
Or maybe something that I can do using KicadStepUp?

luzpaz, 2020-12-22:
Do you mean totally 'headless' (AKA no GUI) or that the program can be open and automated triggered from a script/macro? If the forumer, I guess this would require that all or some of the components used in this workbench be able to function while the GUI is up `FreeCAD.GuiUp()`

leoheck, 2020-12-27:
I was thinking to have something automated (as in no hands). So, any kind of initial automation that do not require user interference (using the mouse, and clicks), however being configurable by the command line would be nice too.
For instance, something you can add inside a script, Makefile. Because this is somehow automation that most of the time do not need interaction, and it can be one of the deliverables of the ECAD with the GERBERs, the PDF of the circuit, the 3d model of the board, and these rendered images showing the final/expected board.

luzpaz, 2021-10-15:
@easyw ?