easyw/kicadStepUpMod #64: Crash on Add Tracks

berkakinci, 2020-11-26:
When I add tracks, I get a FreeCAD crash. (Yes; this may be a FreeCAD issue).
I tried isolating the responsible features in the kicad-pcb file, but I was unsuccessful.  Without isolation, I can't share the whole design.
Basically, it first crashed after `done copper layer F.Cu`.  I tried lopping features out of B.Cu until it didn't crash.  I finally got it not crashing.  However, I had a string of no-crash imports _even after_ I had replaced every single feature back on the board.  Then, after a few imports, it started crashing again.  I don't believe a specific feature is to blame.

Versions:
```
kicad StepUp version 9.7.1.0

OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23074 (Git)
Build type: Release
Branch: master
Hash: 74b866ed3022fd1452c534dae31a721bcc668f2f
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
```

Session log:
```
FreeCAD 0.19, Libs: 0.19R23074 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2020
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

A3 available
KiCadStepUpWB.Activated(v 10.1.1)
PoM not present
using 'Part' container and 'Links'
FC Version 019-23074
Asm3 WB present
kicad StepUp version 9.7.1.0
tolerance on vertex applied
applying Materials to Shapes
your home path is /Users/berka
export to STEP False
Tabified done !
done!
opening /Users/berka/Documents/QCM/Design - Electrical/Schematic/Discovery-Q/Discovery-Q_Plate/Rev02/Discovery-Q-Plate.kicad_pcb
my file path /Users/berka/Documents/QCM/Design - Electrical/Schematic/Discovery-Q/Discovery-Q_Plate/Rev02
aux origin used: [200, 150]
PCB Loader 
pcb thickness 1.6mm
kicad_pcb version 20171130
parsing
16 edge segments
aux_axis_origin[200, 150]
grid_origin[200, 150]
2d circle closed path
2d circle closed path
2d circle closed path
2d circle closed path
2d circle closed path
parsing&building time 0.106
start adding constraints to pcb sketch
adding constraints time 0.425
created PCB face w/ edge tolerance -> 0.01 mm
running time: 1.068sec
start cutting
[... 1155 lines cut ...]
mapped file                       85.9M       51 
shared memory                      652K       18 
===========                     =======  ======= 
TOTAL                              3.0G     2331 
TOTAL, minus reserved VM space     2.3G     2331 

Model: MacBookAir9,1, BootROM 1037.147.4.0.0 (iBridge: 17.16.16610.0.0,0), 4 processors, Quad-Core Intel Core i5, 1.1 GHz, 16 GB, SMC 
Graphics: kHW_IntelIrisPlusGraphicsItem, Intel Iris Plus Graphics, spdisplays_builtin
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, LPDDR4X, 3733 MHz, SK Hynix, H9HCNNNFBMALPR-NEE
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, LPDDR4X, 3733 MHz, SK Hynix, H9HCNNNFBMALPR-NEE
AirPort: spairport_wireless_card_type_airport_extreme, wl0: Feb 28 2020 15:26:30 version 16.20.192.27.3.6.77 FWID 01-f80dbb9b
Bluetooth: Version 7.0.6f7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: USB 3.1 Bus
USB Device: Apple T2 Bus
USB Device: Touch Bar Backlight
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 80.0

```

berkakinci, 2020-11-28:
I think I've pretty much confirmed this is a FreeCAD v0.19 issue.  I was not able to reproduce on v0.18.4.
Can you provide any insight into how FreeCAD may be stumbling into this crash?  I'm hoping you have a better sense of when this may be happening.  I'd like to provide some more context for a FreeCAD bug.

NOT reproducible on:
```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16146 (Git)
Build type: Release
Branch: (HEAD detached at 0.18.4)
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

easyw, 2020-11-28:
@berkakinci 
It would be something related to FC (Qt) version and your mac drivers or even related to OCC version...
I'm sorry but my mac has gone so I cannot test in mac anymore...
Moreover not having the board is it even quite difficult to find the culprit

patfelst, 2020-12-01:
I have just downloaded FreeCAD_0.19-23141 for Mac, and have the same issue. I went to v0.19 because the workbench list is not updating in 0.18.

So in v0.19 the board and components load up, but crashes when loading the tracks.

berkakinci, 2020-12-01:
I'll ask about this in the FreeCAD forum, and hopefully move this info to a bug.

berkakinci, 2020-12-01:
The forum entry is: [https://forum.freecadweb.org/viewtopic.php?f=3&t=52775](url)

berkakinci, 2020-12-02:
> So in v0.19 the board and components load up, but crashes when loading the tracks.

@patfelst Do you have a kicad-pcb design file you can share?  I can't share mine, and it appears without an input file, this bug will meet resistance from devs.

patfelst, 2020-12-02:
I also can't share my design, but I'm sure it will crash for any PCB. I will try and get a chance to create a small mock up PCB and share it. Might not be for a couple of days.

berkakinci, 2020-12-03:
> I also can't share my design, but I'm sure it will crash for any PCB.

You are _absolutely_ right.  I don't know why I didn't think about it earlier, but it happens on the demo.kicad-pcb too.

In the past few days, there was an update.  It now says "KiCadStepUpWB.Activated(v 10.1.4)."  This doesn't agree with the help text, but whatever...

```
...
done copper layer Front
Process 87028 stopped
* thread #1, name = 'CrBrowserMain', queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=EXC_I386_GPFLT)
    frame #0: 0x00000001015a8070 libFreeCADApp.dylib`App::DocumentObject::getNameInDocument() const + 16
libFreeCADApp.dylib`App::DocumentObject::getNameInDocument:
->  0x1015a8070 <+16>: testb  $0x1, (%rax)
    0x1015a8073 <+19>: jne    0x1015a807e               ; <+30>
    0x1015a8075 <+21>: incq   %rax
    0x1015a8078 <+24>: popq   %rbp
Target 0: (freecad) stopped.
(lldb) bt
* thread #1, name = 'CrBrowserMain', queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=EXC_I386_GPFLT)
  * frame #0: 0x00000001015a8070 libFreeCADApp.dylib`App::DocumentObject::getNameInDocument() const + 16
    frame #1: 0x00000001003ae1f2 libFreeCADGui.dylib`Gui::DocumentItem::getViewProvider(App::DocumentObject*) + 50
    frame #2: 0x00000001003a6396 libFreeCADGui.dylib`Gui::DocumentItem::populateItem(Gui::DocumentObjectItem*, bool, bool) + 934
    frame #3: 0x00000001003a30fd libFreeCADGui.dylib`Gui::DocumentItem::createNewItem(Gui::ViewProviderDocumentObject const&, QTreeWidgetItem*, int, std::__1::shared_ptr<Gui::DocumentObjectData>) + 941
...
```

berkakinci, 2020-12-03:
@patfelst
Hopefully you're already past this...  Just in case: I have both 0.18.4 and 0.19_pre in my /Applications (MacOS).  I found that after installing kicadStepUp in 0.19_pre, and opening 0.18.4, the add-on was already there; no install needed specific to 0.18.4.
This makes me more nervous about my practice of running multiple versions.  I haven't been bitten badly yet.

easyw, 2020-12-03:
Addon (WBs) are installed on FC user's profile ... then Addons are shared between different releases of FC 
ATM FC doesn't have a way to place WBs in different folders for different releases (like stable and nightly)
All external WBs are installed in: 
UserAppData | Path where FreeCAD stores User Related application data
https://wiki.freecadweb.org/Start_up_and_Configuration

berkakinci, 2020-12-18:
Hi @easyw,
Do you want to keep this bug here now that it's more readily reproducible with this project?  I do think it really belongs with FreeCAD, but if you're willing to dig in, you may be able to give better insight for how this happens.
I saw in the FC forum he last time there was a crash in App::DocumentObject::getNameInDocument() wmayer jumped in and made a checkin to make the problem go away.  He didn't elaborate.

easyw, 2020-12-20:
@berkakinci is that solved for you with the wmayer patch?

berkakinci, 2020-12-20:
@easyw Sorry for the confusion.  The issue and fix from wmayer was 2017; so no; it predates this issue.
There's a handful of hits in the forum with this function in the stack, but mostly older posts.

berkakinci, 2020-12-20:
I just tried this on the latest version with demo PCB.  It still happens.  It's not every time.  It didn't crash on my first "add tracks."  It's crashed on 3 other test since.
I have reproduced it running under lldb, and I've dumped a core file.  However, it's 4.0GB and still growing.  I'll keep it around if someone is brave enough to dig in to it.

```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23323 (Git)
Build type: Release
Branch: master
Hash: 512d5c6141aec52b6eecc67370336a28fde862a6
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
```

berkakinci, 2021-04-09:
Another user reported the issue in the forums with release version of 0.19.
Since my initial report, I do think the crash is FreeCAD's fault, and not specific to StepUp.
Any idea how to easily transfer this over to FreeCAD?  It may be better received coming from a contributor than me.

easyw, 2021-04-19:
@berkakinci 
I'm sorry I don't have a mac pc anymore... so I can't help further

berkakinci, 2022-01-08:
I've opened the related issue in FreeCAD:
[https://tracker.freecad.org/view.php?id=4823](url)

luzpaz, 2022-04-30:
Per wwmayer in https://tracker.freecad.org/view.php?id=4823#c16201

> Fixed with https://github.com/FreeCAD/FreeCAD/commit/771d9a735abed245e6b095dc1408df68b7d7a3cc and https://github.com/FreeCAD/FreeCAD/commit/29ae8b0502d34946b776cd6fcea43e1d085f857d

CC @berkakinci please close issue if indeed solved.

luzpaz, 2023-03-06:
@berkakinci ping

luzpaz, 2023-04-24:
@easyw can this be closed?

berkakinci, 2023-04-24:
Yes; This can be closed.
I just tried it 5-or-6 times with no crash.
I know; It's not much of a test for something that had a very random symptom.  However, with wmayer's identification of fix, I'm satisfied.
@easyw: There was a workaround you implemented to avoid the problem.  In case it's still there, you may want to remove it.

closed 2023-04-24 by berkakinci