easyw/kicadStepUpMod #65: Not loading 3D components with KiCad v5.99 file format

patfelst, 2020-11-29:
I'm using KiCad V5.99.0-7201 which is a nightly build. When loading a board in FreeCad v0.18 16146 none of the 3D components are loaded onto the board.

Also copper fill zones are not loaded, on basic tracks and vias are loaded.

MitjaNemec, 2020-11-29:
might be connected to #66 and [KiCad s-expr keyword change](https://gitlab.com/kicad/code/kicad/-/commit/3f1a58bd068d20a2008082532996b48d36e261e5). Try editing kicad_pcb to replace all keywords `footprint` with `module` as a temporary workaround. Obviously make a backup first

MitjaNemec, 2020-11-29:
Or you might try changing line 9587 in kicadStepUptools.py from:
`model_name = re.search(r'\(module\s+(.+?)\(layer', model, re.MULTILINE|re.DOTALL).groups(0)[0]`
to
`model_name = re.search(r'((\(module\s)|(\(footprint\s))+(.+?)\(layer', model, re.MULTILINE|re.DOTALL).groups(0)[0]`

patfelst, 2020-11-29:
The python route looks easier, I will try tomorrow, thank you.

easyw, 2020-11-29:
@patfelst 
as I suggested here https://github.com/easyw/kicadStepUpMod/issues/66#issuecomment-735431427 for pcb & 3d models the parser is different... it would be more work to fix it

MitjaNemec, 2020-11-29:
@patfelst As already mentioned by easyw, some additional modifications are required. As it is getting late here I don't have the power to reason in depth much less test my solution and it does not seem likely I'll be able to test it in next couple of days you might want to try it.

On line 1615 in "kicadStepUptools.py" change `('module',` to `('module','footprint'`

easyw, 2020-11-30:
@easyw @MitjaNemec 
I just discovered that @realthunder fixed this issue here https://github.com/realthunder/fcad_pcb/commit/b67059d87aaf42f7b0bca4236d3a633abac6d16e
That should be ported to KSU to fix our issue

patfelst, 2020-11-30:
Sounds promising. I'm not familiar with the code and releases, when and how would this be "ported to KSU". Does it need to be compiled or could I copy the python code in the two files in the fix?

easyw, 2020-11-30:
@patfelst @MitjaNemec it should be now fixed...
please test and confirm.

patfelst, 2020-12-01:
It's working perfectly now - thanks! Even the copper fills are working, as are pads and vias. 

In fact I think there has been an improvement as previously, the solder mask was hiding copper fills, and I had to move those object to above the solder mask to make them visible.

closed 2020-12-01 by patfelst