easyw/kicadStepUpMod #68: Imported STEP file, pins projected in odd location.

robot-army, 2020-12-17:
Hi, I've imported a step file (attached) and attempted to use this tutorial to extract the pin locations from the model. 

When I try to project the pins (2D object to 3D projection) they show up in a very odd place. Sometimes in the correct orientation, but never overlaid on the plane of the pins. See screenshot and freecad file. 
[i7C_Base_unit_FullFeature.zip](https://github.com/easyw/kicadStepUpMod/files/5706611/i7C_Base_unit_FullFeature.zip)

I've also uploaded a screenshot. 

[image]

Thanks

easyw, 2020-12-18:
@robot-army 
you need to place your model aligned to top view and after that project the pins.

easyw, 2020-12-20:
@robot-army I'm closing it... feel free to reopen it in case you need.

closed 2020-12-20 by easyw