easyw/kicadStepUpMod #69: Kicad_mod output

robot-army, 2020-12-17:
The attached FreeCAD file generates the below Kicad_MOD file. When opening it in KiCAD, it fails with an error (which is on a slightly different bit of text) due to the space in the pathname. 

Deleting the 'model' section fixes it. 

[i7c_fromscratch.zip](https://github.com/easyw/kicadStepUpMod/files/5707955/i7c_fromscratch.zip)

[image]

```(module Group (layer F.Cu) (tedit 5A74E519)
  (descr "Grou StepUp generated footprint")
  (fp_text reference "REF**" (at 0.0 -17.0) (layer F.SilkS)
  (effects (font (size 1.0 1.0) (thickness 0.150)))
  )
  (fp_text value "Value " (at 11.0 9.0) (layer F.SilkS)
    (effects (font (size 0.8 0.8) (thickness 0.120)))
  )
  (fp_text user %R (at 0.0 -17.0) (layer F.Fab)
    (effects (font (size 1.0 1.0) (thickness 0.150)))
  )
  (fp_text user "EDIT PAD NUMBERS" (at 11.0 8.0) (layer Cmts.User)
    (effects (font (size 1 1) (thickness 0.2) italic))
  )
  (pad # thru_hole circle (at 2.5 -26.04) (size 2.5 2.5) (drill 1.3) (layers *.Cu *.Mask))
  (pad # thru_hole circle (at 2.5 -18.42) (size 2.5 2.5) (drill 1.3) (layers *.Cu *.Mask))
  (pad # thru_hole circle (at 2.5 -10.8) (size 2.5 2.5) (drill 1.3) (layers *.Cu *.Mask))
  (pad # thru_hole circle (at 14.57 -26.04) (size 2.5 2.5) (drill 1.3) (layers *.Cu *.Mask))
  (pad # thru_hole circle (at 18.380000000000003 -26.04) (size 2.5 2.5) (drill 1.3) (layers *.Cu *.Mask))
  (pad # thru_hole circle (at 30.44 -26.04) (size 3.4 3.4) (drill 1.84) (layers *.Cu *.Mask))
  (pad # thru_hole circle (at 30.44 -22.23) (size 2.5 2.5) (drill 1.3) (layers *.Cu *.Mask))
  (pad # thru_hole circle (at 30.44 -18.42) (size 2.5 2.5) (drill 1.3) (layers *.Cu *.Mask))
  (pad # thru_hole circle (at 30.44 -14.610000000000001) (size 2.5 2.5) (drill 1.3) (layers *.Cu *.Mask))
  (pad # thru_hole circle (at 30.44 -10.8) (size 3.4 3.4) (drill 1.84) (layers *.Cu *.Mask))

   (model C:\Users\robotarmy\red acted\Bluesky\Heater PSU\KiCAD project\heater pcb\TDK_i7c.pretty\Grou.wrl
    (at (xyz 0 0 0))
    (scale (xyz 1 1 1))
    (rotate (xyz 0 0 0))
  )
)
 ```

easyw, 2020-12-18:
@robot-army 
thanks for reporting it ... it is related to the saving path which includes 'spaces' .. I'm going to fix it.
M

easyw, 2020-12-25:
@robot-army 
it should be fixed now.

closed 2021-01-04 by easyw