easyw/kicadStepUpMod #72: Footprint and model not aligned on a plane

robot-army, 2020-12-23:
Hi! I don't know how to describe the issue I'm having using words, so here's a video: 

https://youtu.be/Efx-EG6Ullc

I guess the question is: I've imported a STEP file, how do I move it so that when I make and export sketches to define a footprint, they re-import in such a way that they're aligned (and I don't need to manually push things around by mm increments to line them up?)

easyw, 2020-12-25:
@robot-army 
you don't need to create a Body using the feature as internal part... simply create a sketch on x-y plane and use this sketch to create your fp. You need to have a sketch viewed from top view to maintain the right placement.

robot-army, 2020-12-25:
I don't think I was able to project the edges of the pins into the sketch
if the step import wasn't a base feature of the same body. Anyways, using
(another one of your other awesome contributions to FreeCAD) the
manipulator WB, I was able to make a workflow that made sense to me.

I plan on shooting a video of that workflow sometime before the end of the
year and posting it up. Would love to work with you on it to get it really
correct, if you have time?

Planning on 'STEP to KiCAD two ways', one way is starting from sketcher wb
and the dodgy datasheet dimensions, make the KiCAD_MOD file, and then align
step file using manipulator wb...the other would be the way I've done it so
far, projecting the pins etc.

Thanks! Merry Christmas!!

easyw, 2020-12-26:
@robot-army 
please have a look at the following video:
1) rotate the STEP model to be seen from Top view
2) align the model to center of axis
3) reset the placement for the model
4) get a sketch of the base (seen from bottom)
5) return to top view and get a sketch copy of the inverted sketch, but with the right orientation (top view)
6) then just create your pads and silk screen :smile: 

https://user-images.githubusercontent.com/3032347/103155415-08a06180-47a0-11eb-9640-d24d49c18e75.mp4

luzpaz, 2023-03-06:
Can this ticket be closed ?

closed 2023-04-24 by easyw