easyw/kicadStepUpMod #74: Improvement for Load Footprint tool-tip

leoheck, 2021-01-23:
Hi, @easyw is it possible to improve the tooltip that loads kicad footprint 
from:  `Load Kicad PCB Footprint` 
to: `Load Kicad Footprint`

The `PBC` there bugs me all the time.

luzpaz, 2022-05-11:
@easyw care to weigh-in?

leoheck, 2022-05-12:
It bugs me because there is another tool that says `Load Kicad PCB`, so, the load footprint could be `Load Kicad Footprint`. Also, since the KicadStepUp *is for Kicad (only)* as it is being stated on its name, those tooltips could also drop the `Kicad` string. So it could be:

- `Load PCB`
- `Load Footprint`

Well, to help further, it could have the supported extension in parentheses like:

- `Load PCB (*.kicad_pcb)`
- `Load Footprint (*.mod or *kicad_mod)`

closed 2022-07-03 by easyw