easyw/kicadStepUpMod #75: "Push 3D moved models to PCB" in KiCad 5.99 error due to time stamp change

pjx3, 2021-02-18:
As discussed in this thread: https://forum.kicad.info/t/kicad-stepup-cant-push-model-move-back-to-pcb/27765

The following error is reported when using "Push 3D moved models to PCB" option...

select only 3D model(s) moved to be updated/pushed to kicad board!
a Time Stamp is required!

instead of bringing up the file dialog.

Also, I've noticed that the "Sync 3D model(s) Ref & TimeStamps" option fails to find a part, probably related to the same change within KiCad?

easyw, 2021-02-20:
@pjx3 could you please test the latest commit? it should be fine now also for kicad dev 5.99

pjx3, 2021-02-20:
hi @easyw just tested it and confirmed it's working. many thanks!

closed 2021-02-20 by easyw