easyw/kicadStepUpMod #79: Pad on antenna footprint doesn't import to FreeCAD

kharar, 2021-03-23:
When importing tracks from pcb file I get an exception message in the Report view of FreeCAD and everything EXCEPT one of the pads gets correctly imported.
[Antenna-test-design.zip](https://github.com/easyw/kicadStepUpMod/files/6188830/Antenna-test-design.zip)

I have tried playing around with the footprint, combining the polygon shape with pad 1 or pad 2 or none

[image]
[image]
[image]

In the end I managed to have obtain a useful result by resizing the polygon to overlap pad 2 by 0,1mm and completely overlapping pad 1 and then combine pad 1 and the polygon - though I believe this is only a workaround as pad 1 has just been overlapped and is still missing from the import.
[image]

Am I doing something wrong or is this a bug in the "ksu tools Add tracks" code?

Here is the exception message that I get:

> kicad StepUp version 9.7.3.5
> 11:57:30  tolerance on vertex applied
> 11:57:30  opening C:/Users/Martin/Documents/KiCad/Antenna-test-design-for-Cenos/Antenna-test-design-for-Cenos.kicad_pcb
> 11:57:30  my file path C:\Users\Martin\Documents\KiCad\Antenna-test-design-for-Cenos
> 11:57:30  grid origin found @ (-100, 100)
> 11:57:30  PCB Loader 
> 11:57:30  pcb thickness 1.6mm
> 11:57:30  kicad_pcb version 20171130
> 11:57:30  parsing
> 11:57:30  4 edge segments
> 11:57:30  aux_axis_origin[160, 100]
> 11:57:30  grid_origin[100, 100]
> 11:57:30  parsing&building time 0.002
> 11:57:30  start adding constraints to pcb sketch
> 11:57:30  using constrainator
> 11:57:30  check to sanitize
> 11:57:30  adding constraints time 0.056
> 11:57:30  created PCB face w/ edge tolerance -> 0.01 mm
> 11:57:30  running time: 0.078sec
> 11:57:30  start cutting
> 11:57:30  using hierarchy container
> 11:57:30  cutting time 0.0
> 11:57:30  pcb dimensions: (60.00;16.00;1.60)
> 11:57:30  running time: 0.086sec
> 11:57:30  FC Version 019-24276
> 11:57:30  creating hierarchy
> 11:57:30  placing board @ -100,100
> 11:57:30  building up pcb time 0.087
> 11:57:30  running time: 0.117sec
> 11:57:30  VBO status False
> 11:57:30  ReadShapeCompoundMode status True
> 11:57:30  ${KISYS3DMOD}/Connector_Coaxial.3dshapes/U.FL_Hirose_U.FL-R-SMT-1_Vertical.wrl
> 11:57:30  adjusting Local Path
> 11:57:30  step-module-replaced Connector_Coaxial.3dshapes/U.FL_Hirose_U.FL-R-SMT-1_Vertical.wrl
> 11:57:30  model name U.FL_Hirose_U.FL-R-SMT-1_Vertical
> 11:57:30  opening C:/Program Files/KiCad/share/kicad/modules/packages3d\Connector_Coaxial.3dshapes/U.FL_Hirose_U.FL-R-SMT-1_Vertical.step
> 11:57:30  U.FL_Hirose_U.FL-R-SMT-1_Vertical
> 11:57:30  <Import> ImportOCAF2.cpp(541): free shape count 1
> 11:57:30  module Connector_Coaxial.3dshapes/U.FL_Hirose_U.FL-R-SMT-1_Vertical.step
> 11:57:30  added 1 model(s)
> 11:57:30  running time: 0.265sec
> 11:57:30  info message
> 11:57:30  Board Placed @ -100.00;100.00;0.0
> 11:57:30  kicad pcb pos: (130.00;-92.00;0.00)
> 11:57:30  pcb dimensions: (60.00;16.00;1.60)
> 11:57:30  thread ViewFitting
> 11:57:32  running time: 0.265sec
> 11:57:32  FreeCAD build date: 2021/03/11
> 11:57:32  STEP UseAppPart available
> 11:57:34  tracks version: 1.1
> 11:57:36  making copper layer F.Cu...
> 11:57:36      making holes...
> 11:57:36        pad holes: 0, skipped: 0
> 11:57:36        oval holes: 0
> 11:57:36        via holes: 0, skipped: 0
> 11:57:36        total holes added: 0
> 11:57:36      holes done
> 11:57:36    making pads...
> 11:57:36  polyline points: 13
> 11:57:36      modules: 2
> 11:57:36      pads: 6, skipped: 0
> 11:57:36      vias: 0, skipped: 0
> 11:57:36      total pads added: 6
> 11:57:36    pads done
> 11:57:36    making tracks...
> 11:57:36      making 4 tracks of width 1.00, (0/4)
> 11:57:36    tracks done
> 11:57:36    making zones...
> 11:57:36      making zone GND...
> 11:57:36        region 1/1, holes: 0
> 11:57:36    zones done
> 11:57:36  done copper layer F.Cu
> 11:57:36  making copper layer B.Cu...
> 11:57:36      making holes...
> 11:57:36        pad holes: 0, skipped: 0
> 11:57:36        oval holes: 0
> 11:57:36        via holes: 0, skipped: 0
> 11:57:36        total holes added: 0
> 11:57:36      holes done
> 11:57:36    making pads...
> 11:57:36      modules: 2
> 11:57:36      pads: 6, skipped: 6
> 11:57:36      vias: 0, skipped: 0
> 11:57:36      total pads added: 0
> 11:57:36    pads done
> 11:57:36    making tracks...
> 11:57:36    tracks done
> 11:57:36    making zones...
> 11:57:36    zones done
> 11:57:36  <Exception> no shape added
> 11:57:36  Running the Python command 'ksuToolsAddTracks' failed:
> Traceback (most recent call last):
>   File "C:\Users\Martin\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicadStepUpCMD.py", line 3215, in Activated
>     tracks.addtracks()
>   File "C:\Users\Martin\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\tracks.py", line 292, in addtracks
>     pcb.makeCopper(holes=True, minSize=minSizeDrill)
>   File "C:\Users\Martin\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicad_parser.py", line 1423, in makeCopper
>     obj = self._makeArea(objs,'copper',fit_arcs=fit_arcs)
>   File "C:\Users\Martin\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicad_parser.py", line 683, in _makeArea
>     ret = ret.getShape()
> 
> no shape added
>

easyw, 2021-04-09:
> Am I doing something wrong or is this a bug in the "ksu tools Add tracks" code?

there is a bug in the add tracks routine; the import footprint tool is instead accurate.

kharar, 2021-04-09:
That's a shame because it means that you cannot smoothly import PCB's containing said features.
Hoping a fix will be in the pipeline sometime - otherwise one could also argue that the issue is with the way that KiCAD 5.x footprint editor handles footprint pads and copper polygons. Maybe it's about time for me to pay the 5.99 nightlies a visit... ;-)

easyw, 2021-04-12:
> That's a shame because it means that you cannot smoothly import PCB's containing said features

StepUp is aimed to MCAD checking and designing, not on a full trace importing tool.
For a tight traces importing in FreeCAD you may rely on [fcad_pcb](https://github.com/realthunder/fcad_pcb) (which engine is partially used in StepUp)
You can also open an issue at this repo.

closed 2023-03-30 by easyw