easyw/kicadStepUpMod #80: need to discretize Arcs

dom11990, 2021-03-29:
I have a microstrip structure that uses some curves. I know this is not supported by KiCAD so I was hoping StepUp could help in the automatic conversion from arc to line segment with N parts. I'm not very versed in FreeCAD, I'm guessing there is a way to do this...

Currently when I try to export my structure to a footprint I get the following error:

```
10:43:33  saveSettings Helper
10:43:33  PoM not present
10:43:33  using 'Part' container and 'Links'
10:43:33  FC Version 019-24276
10:43:33  kicad StepUp version 9.7.3.5
10:43:33  tolerance on vertex applied
10:43:33  applying Materials to Shapes
10:43:33  your home path is C:\Users\Dominik
10:43:33  kicadStepUp docked
10:44:56  KiCadStepUpWB.Activated(v 10.1.8)
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  need to discretize Arcs
10:45:06  ['REF**']
10:45:06  ['Value ']
10:45:06  Running the Python command 'ksuToolsFootprintGen' failed:
Traceback (most recent call last):
  File "C:\Users\Dominik\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicadStepUpCMD.py", line 1920, in Activated
    kicadStepUptools.PushFootprint()
  File "C:\Users\Dominik\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicadStepUptools.py", line 16575, in PushFootprint
    sk_d=Draft.makeSketch([wns,wnc[0]])

list index out of range
```

Seems like the discretization is not handled automatically. Could someone point me in the right direction on how this is done? Or Is this error related to something completely different? I can provide the file if it helps.

easyw, 2021-03-29:
@dom11990 there is a button to discretize bsplines to arcs or lines... would you mind to post (zipped) your Fc file?

dom11990, 2021-03-30:
@easyw So I've written a python script to handle this but it would be interesting to know if it is supported in KSU. I've attached the dxf and below is a picture of what I would like the footprint to look like after all is said and done. Essentially, I need to "fill" the area enclosed by my DXF file. I was not able to figure out how to do this.

[image]

[wilkinson.zip](https://github.com/easyw/kicadStepUpMod/files/6227140/wilkinson.zip)

easyw, 2021-03-30:
I have replied at kicad.info forum
https://forum.kicad.info/t/new-way-to-import-dxf-outlines-to-footprint/28516/5

dom11990, 2021-03-31:
[wilkinson.zip](https://github.com/easyw/kicadStepUpMod/files/6234400/wilkinson.zip)

This is the final result I would like as a footprint:
[image]

easyw, 2021-04-09:
it should be solved with latest commits
[Commits on Apr 8, 2021](f39f8983498e2eadbac9d85fd1ced03ad9eee7ca)

dom11990, 2021-08-08:
Just tried it out, works great! Runs much faster than doing it in Kicad like I did with the action plugin. Thanks for this awesome feature!

closed 2021-08-08 by dom11990