easyw/kicadStepUpMod #84: Feedback issue converting FreeCad STEP to KiCad Footprint - Python Traceback and excessive RAM

dewezcyril, 2021-04-26:
Dear all!

Firstly I don't know how to get in touch with you apart from this GitHub issue section. I apologize if you don't like it this way.

I used KiCad to convert a STEP 3D design to a flat KiCad PCB footprint as explained in this tutorial ([generating a kicad footprint from a 3D STEP model - YouTube](https://www.youtube.com/watch?v=5IcAyfhaf2g)). I got issues at the last step, which is the final export to the kicad_mod file. Either FreeCad uses too much RAM and my PC crash, or I get a Python fail Traceback. The console prints are copy-pasted below for your debugging.
The STEP file I wanted to convert can be found here ([link to the download](http://alchemy-power.com/wp-content/uploads/2018/10/Pi-16ADC-121516-STEP.zip)).
Versions used were: Windows 10 19041.928, FreeCad 0.19, KiCadStepUp installed via FreeCad addon manager today (23-04-21).

Does one of you know what could be my mistake seen the Traceback below?
On the other hands, I would be really happy if someone could help me converting this STEP to a footprint ;-).

Yours faithfully,
Cyril

```
14:59:35  exporting new footprint
14:59:35  saving to C:/Users/Luc/Desktop/PI16-ADC-footprint-v1.kicad_mod
14:59:35  Group
14:59:35  Sketch
14:59:35  Text
14:59:35  Text001
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  2d closed path
14:59:35  Running the Python command 'ksuToolsFootprintGen' failed:
Traceback (most recent call last):
  File "C:\Users\Luc\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicadStepUpCMD.py", line 1986, in Activated
    kicadStepUptools.PushFootprint()
  File "C:\Users\Luc\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicadStepUptools.py", line 16621, in PushFootprint
    export_footprint(name)
  File "C:\Users\Luc\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicadStepUptools.py", line 17248, in export_footprint
    drl_found=collect_drl(pth_ordered)
  File "C:\Users\Luc\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicadStepUptools.py", line 17887, in collect_drl
    r2=p[2][1]; cx2=p[2][2]; cy2=p[2][3]
```

easyw, 2021-04-27:
@dewezcyril 
what do you want to get from your file?
Do you want a fp with edgecuts and some pads?

easyw, 2021-04-28:
@dewezcyril 
attached an option for creating a footprint from your step model
[Pi-16ADC-121516-STEP-fp.zip](https://github.com/easyw/kicadStepUpMod/files/6390740/Pi-16ADC-121516-STEP-fp.zip)

dewezcyril, 2021-04-28:
@easyw thank you very much for your .zip file! This is exactly what I am looking for. I exported it to kicad_mod successfully.
I forgot the layers F_Silks_0.16 and Edge_Cuts_0.15, I suppose this was the reason why it failed before.

Thank you very much!

closed 2021-05-06 by easyw