easyw/kicadStepUpMod #86: Question / feature request: Exporting KiCad PCB to STEP from the command line / python script

burnpanck, 2021-06-05:
I would like to automatically export 3D models of my PCBs from within our automated PCB build flow. The KiCad native STEP exporter is known to generate wrong output (at least in 5.1.x) whenever the component models are scaled and/or moved within the footprint library, despite the fact that the 3D viewer correctly displaying them. StepUp on the other hand exports like a charm.
Unfortunately, I haven't found a way to run these processes automatically. I have tried to use the macro recording functionality of FreeCAD, but unfortunately, the StepUp commands do not record anything. I do not need macro recording functionality, just some pointers to how this could be done from a python script directly - maybe somebody else has done this before? The functionality is surely contained in `kicadStepUptools.py`. That file just isn't exactly easy to read for somebody unfamiliar with the code-base like me. I would be happy to dig myself once I know where to start.

easyw, 2021-06-14:
@burnpanck 
there is an option in StepUp to enable direct exporting after having 'opened' a kicad pcb in FC.
If this option is checked, the tool will export the imported pcb as 'step' file

AlexColello, 2021-07-25:
@easyw I think you are missing the point of the request. The idea is you don't have to open any program, and just run a python command with a kicad file as the input and a step file as an output. I was looking the ability to do this as well, and would love to see it happen!